gdsfactory/gdsfactory #1553: Better KLayout TilingProcessor scaling

simbilod, 2023-04-14:
PR #1535 added tiled fill, thanks to @sebastian-goeldi and @SkandanC 

Using default parameters and simply changing n_threads results in poor scaling:

Sample cell:

| n_threads      | Script time (s) |
| ----------- | ----------- |
| 1      | real    2m54.031s
| | user    2m55.305s
| | sys     0m13.799s       |
| 5   | real    1m27.213s
| | user    2m55.673s
| | sys     0m13.970s        |
| 10   | real    1m16.487s
| | user    2m56.166s
| | sys     0m13.860s        |

4x larger cell:

| n_threads      | Script time (s)  |
| ----------- | ----------- |
| 10      | real    4m1.849s
| | user    10m41.205s
| | sys     0m13.980s       |
| 20   | real 3m37.944s |
| | user 10m39.971s |
| | sys 0m14.395s |

I've noticed only a small part of the overall script actually launches n_threads.

Is there any advice on how to better improve the parallelization? Are there optimal ways to define fill_layers vs fill_regions and tile_size?

sebastian-goeldi, 2023-04-14:
How much it scales with number of threads entirely depends on your machine and the problems. Tl;dr I can only think of one problem and I am not even sure I left that in the code (the fill cell layers might be in the build part of the exclusion, that might not be smart).

Your profiling contains very little information about your machine and the problem you are trying to solve, so I'll have to make a bunch of assumptions.

* Your machine has more cores available than threads you are using
* You aren't running out of memory (swap doesn't count)
* You are using relatively few layers to calculate the area to fill
* The resulting areas are almost exclusively complex (many points, non-trivial shapes, like doesn't result in rectangles or similar)
* You tried multiple sizes for the tiles.

If any of this is true, I am almost certain that your filling is spending most of the time in the TileOutputProcessor part. That part is not (safely) parallelizable, see https://www.klayout.de/forum/discussion/550/using-the-fill-tool

If you want to dive deeper into this issue, I suggest to use a proper profiler and see where python spends the most time.

sebastian-goeldi, 2023-04-14:
If you can, avoid fill regions and just use layer indexes

simbilod, 2023-04-15:
Thanks Sebastian, this is great advice. Yes you are dead on with all your bullet points, and I do seem to spend most of my time during TileOutputProcessor.

sebastian-goeldi, 2023-04-15:
Ok, in that case your main benefit of the tiling processor comes from splitting the region to be filled. For large regions it can take ages to allocate memory and also to make the region object due to the sheer amount of polygons.

You could also do this manually and it might be faster, but I doubt that tbh.

But as you have seen, such a case doesn't benefit from multi-threading a lot.  This is a classic problem in parallel-computing if you have a work load that is largely dominated by one function that is not further parallelizable (this is also the reason why nowadays a lot of programs and games don't take advantage of the number of cores available in machines).

joamatab, 2023-10-24:
moved to gplugins

closed 2023-10-24 by joamatab