gdsfactory/gdsfactory #1556: tidy3d credentials seem to be missing when building docs

joamatab, 2023-04-17:
tidy3d credentials seem to be missing when building docs

[image]

@tyler-flex
@lucas-flexcompute 
@momchil-flex

joamatab, 2023-04-17:
Where is tidy3d looking for the credentials?

is it still in `~/.tidy3d/auth.json`?

tylerflex, 2023-04-17:
Hi @joamatab 

Yes, the file has moved to `~/.tidy3d/config` and it should contain 
```
apikey = "XXX"
```
where XXX is your API key (and the quotes are surrounding it).

Alternatively (might be easier in GitHub actions) you could set an environment variable of 
```
SIMCLOUD_APIKEY="XXX"
```

There's some more detailed instructions [here](https://docs.flexcompute.com/projects/tidy3d/en/latest/quickstart.html#linking-regiestration)

Let us know if that solves the issue!

joamatab, 2023-06-10:
yes, it solved it, thank you Tyler

closed 2023-06-10 by joamatab