gdsfactory/gdsfactory #1565: triangle does not build on Mac

simbilod, 2023-04-20:
causes pip install gdsfactory[full] to fail on Mac

what is this package doing? Can we remove it? @joamatab

joamatab, 2023-04-20:
triangle is for 3D rendering,

can you try `conda install gdsfactory` first?

joamatab, 2023-04-20:
it seems to be working on CI/CD

https://github.com/gdsfactory/gdsfactory/actions/runs/4757347263/jobs/8454060958

is it an M1/M2 or a regular Intel chip?

joamatab, 2023-04-20:
https://github.com/github/roadmap/issues/528

simbilod, 2023-04-20:
@thomaslima @ericcblow @jz8939

thomaslima, 2023-04-20:
It's an M1 Mac. This is the output of `conda install gdsfactory`:

```
conda install gdsfactory
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: /
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versionsThe following specifications were found to be incompatible with your system:

  - feature:/osx-arm64::__unix==0=0
  - feature:|@/osx-arm64::__unix==0=0
  - gdsfactory -> click -> __unix
  - gdsfactory -> click -> __win

Your installed version is: 0
```

thomaslima, 2023-04-20:
The root issue is here: https://github.com/drufat/triangle/issues/62

joamatab, 2023-04-20:
just released a new version, can you try `pip install gdsfactory[full] --upgrade` ?

closed 2023-04-21 by joamatab

joamatab, 2023-04-21:
thank you Thomas!

How fast is `pip install klayout` on M1 Mac?

Im thinking to build a mamba version that includes all gdsfactory packages, to replace constructor, similart to mamba
https://github.com/mamba-org/mamba

#904

thomaslima, 2023-04-21:
Honestly I haven't timed it. I think about 10 minutes.

PR #1567 (2023-04-20): remove triangle
fixes #1565 

removed triangle dependency, built a new environment and tests seem to pass