gdsfactory/gdsfactory #1573: klive from within docker

Jan-David-Black, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
I'd like to encapsulate most of the gdsfactory functionality (incl. plugins) in a docker container. Despite kweb providing some visuals on the generated gds it would be nice to also have easy access to klayout (as it is right now with the klive server).

**Describe the solution you'd like**
@sebastian-goeldi as detailed (here)[https://stackoverflow.com/questions/24319662/from-inside-of-a-docker-container-how-do-i-connect-to-the-localhost-of-the-mach] it is relatively easy (and potentially platform independent) to let processes inside a container access ports on the host machine. I tried talking to the klive server and that seems to work. The question is how do we transfer the gds file:
1. we can use mount points to save the files transmit the corresponding filepath (will differ depending on os, config etc.) and then load the corresponding file using klive on the host machine.
2. we can transfer the file over the TCP socket that we use for klive anyways -> To me that looks like a cleaner solution, that should be less os dependent. (Might create quite a lot of TCP trafic between host and container though)

What do you guys think?

@joamatab 
@proppy

tvt173, 2023-04-24:
(2) would be a more general solution, since the docker container would need not be running on your local machine...
if using a mounted volume, you assume the client is also the host of the docker container (or at least has access to the same mounted drive)

or the other option is to use a kweb-like interface... but i agree, it would be nice to at least have the _option_ to connect to the full klayout application remotely

sebastian-goeldi, 2023-04-24:
I don't see anything in that stackoverflow answer that actually gives a good solution.

`--network=host` is only available in [linux](https://docs.docker.com/network/host/).

In bridge mode you rely on klive attaching the IP of the host on the docker subnet. This is by no means fixed, you can just re-define that network in the docker config and klive won't work anymore.

Imo the whole concept is backwards. And it completely falls apart for remote execution. For remote execution to work with this concept, you would need to run klive on 0.0.0.0 and open a port in your firewall. Good luck with that in any corporate network. I think my IT and most IT departments would get a heart attack if you'd suggest that to them.

I think the best solution would be to make the kernel that runs the code remote and implement a way for the local installation to retrieve it by TCP and then send it to kweb or klive. I would suggest [jupyter-kernel-gateway](https://github.com/jupyter-server/kernel_gateway)

For the full docker experience I would then go the way of staying in docker. Run kweb in its own container, share a volume between kweb and gdsfactory (and host).

Still, for the standard uni student, I don't think this is the one click solution that is desired.

Jan-David-Black, 2023-04-24:
I am not so sure about the fully networled solution, but the `host.docker.intrenal` address seems to be available on all OSs (when using the flag `--add-host host.docker.internal:host-gateway`). I agree, that that is not a solurion for networked execution. For that one cold think about klive that connects to a service in the container (kind of a klive proxy) over ws. Then the port direction is in order. The klive proxy will then internally expose 8082 so that gdsfactory can send its rendering requests as usual, while they get forwarded over the ws connection.

sebastian-goeldi, 2023-04-24:
The docker docs are pretty clear about it: It's a development flag only https://docs.docker.com/desktop/networking/#i-want-to-connect-from-a-container-to-a-service-on-the-host 

And again, this approach doesn't solve the problem of klive only listening on localhost/127.0.0.1. This will only allow you to connect to services of the host that listen on the docker bridge IP of the host. And it's just an all around bad idea to have klive listen even on the whole 172.16.0.0/12 is just a bad default behavior. It is fine if the user starts this, but not fine for a service like klive, because depending on your local setup, 172.16.0.0/12 might just be used as a local subnet, as it is not strictly reserved for docker (which seems to use 172.17.0.0/24).

Jan-David-Black, 2023-04-24:
I see your criticism. Basically we don't want a service on the host listening to anything outside of itself. How about the "proxy" solution though?

Jan-David-Black, 2023-04-24:
Going to try to implement it later then...

Jan-David-Black, 2023-04-26:
I now used the following method:
use Docker Compose to set up an MQTT broker in parallel to the devcontainer. Have a service inside the container that translates from the tcp requests on port 8082 to mqtt messages. On the host there is a service that subscribes to the mqtt topic ('klive'). It translates back to tcp and requests klayout to display the gds. The file is transferred using a flask file server serving a single endpoint (the gdsfile).
https://user-images.githubusercontent.com/92856893/234450814-51cdfedc-1c5b-4f44-9995-4e3227099acc.mp4

Jan-David-Black, 2023-04-26:
What is a bit dirty about this approach is that we have this additional service running on the host machine. We could also run it as a KLayout macro but it requires some nonstandard libraries (`paho-mqtt` and `requests`). So it might be a bit tricky to get that reliably installed with one click...

Jan-David-Black, 2023-05-01:
I have set up the solution here: https://github.com/Jan-David-Black/gproject
If you have time pease give it a testdrive.
Closing for now

tvt173, 2023-05-02:
cool! thanks for sharing @Jan-David-Black !