gdsfactory/gdsfactory #1578: improve kweb

joamatab, 2023-04-24:
- [ ] 3D visualization
- [ ] Pyspice integration. Similar to [Siliwiz](https://app.siliwiz.com/)
- [ ] database integration

SkandanC, 2023-04-24:
Do we know what kind of 3d Visualization we plan on adding? Something like the to_3d thing? afaik klayout doesnt have a full 3d viewer, unless there's sme salt package for it

joamatab, 2023-04-25:
How about adding https://gdsfactory.github.io/gdsfactory/notebooks/03_layer_stack.html#layerstack

?

Jan-David-Black, 2023-05-01:
What I would also love to see is an endpoint, that shows the "current" layout like klive does. And the option to show the cell hirachy.

Jan-David-Black, 2023-06-15:
@joamatab, @SkandanC, @sebastian-goeldi   I have started to implement something like that here: https://github.com/Jan-David-Black/gdsfactory/tree/current_component_web I am however not so sure how to cleanly load in the new layout, when the file watcher senses a newly created/modified file:
Currently I am doing something in the lines of:
´´´python
def show_new_gds(self, filepath):
        self.url = self.gds_path = filepath
        self.update_gds()

def update_gds(self):
      self.layout_view.active_cellview().layout().clear()
      self.layout_view.load_layout(self.url)
      self.layout_view.max_hier()
´´´
That seems to trip up the layers though... (their ids for turning on/off from the frontend)

sebastian-goeldi, 2023-06-15:
@Jan-David-Black I think you most likely want to do something similar to what klive is doing here https://github.com/gdsfactory/klive/blob/main/klayout/python/klive_server.py#L63 i.e. `self.layout_view.reload(cell_view_index)` the `cell_view_index` you can get by getting the active cell_view with `self.layout_view.active_cellview_index()`

closed 2023-10-24 by github-actions[bot]