gdsfactory/gdsfactory #1591: update to latest pydantic

joamatab, 2023-04-25:
How can we update to latest pydantic version 2.0?

joamatab, 2023-07-18:
I've tried both migrating to the new [V2 models](https://github.com/gdsfactory/gdsfactory/tree/pydantic2) as well as using the [V1 old API](https://github.com/gdsfactory/gdsfactory/tree/pydantic2_v1)

alibillalhammoud, 2023-07-18:
I was looking at the changes pydantic devs made with v2 today. I briefly took a look at gdsfactory and tried running some stuff. This is not everything, but here are things to consider....

- You have already seen about the "bump-pydantic" package on pypi? [if not check it out](https://github.com/pydantic/bump-pydantic#installation). If you run this first it will be helpful do most of the heavy lifting.
- We probably will require `pip install pydantic-extra-types` package because `from pydantic.color import Color` will not work anymore. 
- [LayerStack](https://github.com/gdsfactory/gdsfactory/blob/ca4b50c4518846f3d7bc6cad317a24cb3374bee7/gdsfactory/generic_tech/layer_stack.py#L57) in generic tech: add type hints here and below. e.g.“substrate: LayerLevel =... ” rather than “substrate = …”
- The confusing part for me is we need to do something with the `__get_validators__` whenever we want to validate a normal class (which does not inherit from pydantic.BaseModel). I think we need to replace it with `__get_pydantic_core_schema__` but I may be wrong about this. see [the error](https://docs.pydantic.dev/dev-v2/usage/errors/#schema-for-unknown-type) and [a possible solution](https://docs.pydantic.dev/latest/usage/types/custom/#as-a-method-on-a-custom-type) (just a guess, try `handler(BaseModel)` instead of `handler(str)` as in the example). There are not that many classes for which we must make this change. run `grep -r "__get_validators__"` from the command line and you will see the places we need to change.

joamatab, 2023-07-18:
Hi Ali,

yes the automatic tool worked better

https://github.com/gdsfactory/gdsfactory/pull/1879

thanks for suggesting that

closed 2023-08-30 by joamatab