gdsfactory/gdsfactory #1597: kfactory backend

joamatab, 2023-04-26:
As part of the kfactory backend evaluation we've been thinking about porting all gdsactory pcells into kfactory and then create a new repo just for gdsfactory plugins

if the evaluation is sucessfull the idea would be to keep the kfactory as new gdsfactory geometry core
this repo would be marked as archived

@SkandanC 
@sebastian-goeldi 
@tvt173 
@simbilod

SkandanC, 2023-04-26:
Would this mean writing all PCells in KFactory? That'd be quite difficult since we have a large amount. Anothe issue is kfactory cant do None orientation ports, so electrical components like vias would be difficult.

sebastian-goeldi, 2023-04-26:
@SkandanC we are aware. There are some workarounds / fixes we can implement. See https://github.com/gdsfactory/kfactory/issues/81

simbilod, 2023-04-27:
Since kfactory was built to emulate gdsfactory in syntax, how hard would it be to programmatically translate the pcells?

SkandanC, 2023-04-27:
There are a lot of key features in kfactry that are nt yet implemented like ymix ymax etc. But theoretically, it should be sort of doable. In my lateest kfactory PR, for example, I added an mzi component - which is written in the same way mroe or less as gdsfactory

sebastian-goeldi, 2023-04-27:
ymin/ymax are there,  they just are readonly on the bounding box. If you really want, you can use a special layer to emulate gdsfactory behavior, but this would be until save of the gds. Meaning after a read, the cell behaves differently. Which is definitely not ideal.

joamatab, 2023-04-28:
Can we merge Skandan branch? I also want to send a PR with some improvements

sebastian-goeldi, 2023-04-28:
It's not ready yet. There is e.g. a mix of components & pcells. I think we should decide on a name. I chose PCell due to klayout & digital naming.

sebastian-goeldi, 2023-04-28:
You could make the PR against Skandan's PR though if your changes need parts of his. Otherwise feel free to just make one against main and Skandan can rebase.

@SkandanC what is exactly missing/not finished yet until we can do a review?

sebastian-goeldi, 2023-04-28:
I see

closed 2023-10-24 by github-actions[bot]