gdsfactory/gdsfactory #1603: API documentation is difficult to navigate

DanHickstein, 2023-04-29:
gdsfactory seems to do a great job with documentation in general! But I find navigating the API documentation to be a little difficult. For example, I'm using the the geometry.offset function. I can easily find this function in the [expected place in the source code](https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/geometry/offset.py). And there is a docstring describing how to use the function. 

But, when I go to the [API documentation](https://gdsfactory.github.io/gdsfactory/api.html), I literally cannot find the documentation for this function. Shouldn't this docstring be in the API docs somewhere?

joamatab, 2023-05-01:
Thank you for pointing that out Dan,

I added some more entries to the API documentation

let me know if it's better with this PR https://github.com/gdsfactory/gdsfactory/pull/1606

closed 2023-05-01 by joamatab

DanHickstein, 2023-05-01:
Thanks @joamatab! 

The API docs seem similar though. For example, I still don't see the "geometry.offset" function listed.

On the main page, the "Sphinx docs to GH pages" badge says "Failing", so perhaps there is some issue with synchronizing to GH pages? 

[image]

joamatab, 2023-05-02:
can you check again? yes, it takes 20 min to build the docs, and during those 20min the workflow shows up as failing :)

https://gdsfactory.github.io/gdsfactory/api.html

DanHickstein, 2023-05-02:
It works now! Thanks :)