gdsfactory/gdsfactory #1607: Release fails

Jan-David-Black, 2023-05-02:
**Describe the bug**
The release action fails. It seems to have problems accessing the repository. 

**To Reproduce**
See Actions

**Expected behavior**
Exit successfully

**Suggested fix**
No idea, will do some digging.

Jan-David-Black, 2023-05-02:
Seems like the Sha is somehow messed up: Sha of the commit:
f6b442a9cd56caa0e6f4bf1a062875ccd45ed2eb

Sha that is referenced by the action:
d0b9f9a7e1d16d57a81b57b8870dcf41bf45bd29

It also doesn't list the tag v6. 90.9 so there might be a race condition, where the repo is not up to date when the action is run?

joamatab, 2023-05-11:
Hi Jan David,

There is an issue with constructor
https://github.com/gdsfactory/gdsfactory/actions/runs/4941954535/jobs/8835052917

@jaimergp

closed 2023-06-10 by joamatab