gdsfactory/gdsfactory #1646: Avoid given areas in `get_bundle`

elamdf, 2023-05-15:
I am trying to route electrical pads on a testbench with grating couplers (which metal layers can't cross over) with `get_bundle`. Is there any way to have the router avoid them by specifying their bounding boxes? I know I can set waypoints, but I'm not sure how that would work with a heterogenous test site (ie the pads I need to route to are not just an array).

elamdf, 2023-05-22:
It appears that `get_route_astar` is what I'm working on. I am having some trouble getting it to halt, but will investigate that further

joamatab, 2023-05-22:
Hi Elam,

i think you are looking for a combination of `gf.routing.route_ports_to_side` followed with `gf.routing.get_bundle`
i can add an example, 

Also,If you can reproduce what you are trying to do using our generic pdk i can give it a try

elamdf, 2023-05-27:
I think this could work, but I think we will have to manually route for
this tapeout as submission deadline is approaching- I'm sure you know the
situation well :) To be continued!

closed 2023-10-21 by github-actions[bot]