gdsfactory/gdsfactory #1647: Issue on page /notebooks/sax/01_sax.html

joamatab, 2023-05-16:
SAX simulations seem wrong after updating tidy3d mode solver with conformal mesh

[image]

https://gdsfactory.github.io/gdsfactory/notebooks/sax/01_sax.html#id1

@flaport 
@SkandanC

lucas-flexcompute, 2023-05-17:
I believe the problem is that we're solving each wavelength in a different grid. Because of the grid differences, each computation of the effective index will have a different quantization error. Here's what I mean:

```python
wavelength = numpy.linspace(1.5, 1.6, 10)
n0 = []

for lda in wavelength:
    wg = waveguide.RectangularDielectric(
        wavelength=lda,
        core_width=0.5,
        core_thickness=0.22,
        core_medium=td.Medium(permittivity=3.48**2),
        clad_medium=td.Medium(permittivity=1.0),
        box_medium=td.Medium(permittivity=1.45**2),
    )
    n0.append(wg.n_eff)

wg1 = waveguide.RectangularDielectric(
    wavelength=wavelength,
    core_width=0.5,
    core_thickness=0.22,
    core_medium=td.Medium(permittivity=3.48**2),
    clad_medium=td.Medium(permittivity=1.0),
    box_medium=td.Medium(permittivity=1.45**2),
)

fig, ax = pyplot.subplots(1, 1)
ax.plot(wavelength, [n.sel(mode_index=0).item() for n in n0], label="Independent")
ax.plot(wavelength, wg1.n_eff.sel(mode_index=0).values, label="Group")
ax.set_xlabel("Wavelength")
ax.set_ylabel("Effective index")
ax.legend()
```
[image]

Unfortunately, the waveguide plugin in gdsfactory accepts only a single wavelength, not a range, so we end up with noisy data that is not appropriate for interpolation with jax/sax. I implemented like this because the previous version was also single-wavelength, if I'm not mistaken.

The best solution seems to be to change the waveguide plugin in gdsfactory to accept multiple wavelengths in a single solver run (which might also be a little more efficient). The only issue I'm not sure how to reconcile is to get the material indices including dispersion. What we do now:
```python
n_core = get_material_index(self.core_material, self.wavelength)
core_medium = td.Medium(permittivity=n_core**2)
```
because the material parameter can be passed as a name. Maybe we could accept a tidy3d material directly? Is there a way to get the tidy3d material directly from the name (maybe something like `get_tidy3d_material(str)`)?

SkandanC, 2023-05-17:
Yep, get_material_index​ can take a material as a str as well. Underneath it just gets the material from tidy3d anyway.

[comparison]<https://user-images.githubusercontent.com/119979961/239063873-bb96f824-361f-46a9-b572-9a9c9d6625d0.png>

lucas-flexcompute, 2023-05-18:
Yes, I understand that. What I meant is to get back a `tidy3d.Medium` object instead of a number, so that dispersion is properly modeled.

SkandanC, 2023-05-18:
Yep, you can use get_medium instead then​

closed 2023-05-29 by joamatab