gdsfactory/gdsfactory #1656: improve `plot_netlist()`

joamatab, 2023-05-17:
How can we make `plot_netlist` go from here

[image]

to there

[image]

tvt173, 2023-05-18:
If you just want to display the bokeh plot, I think you need to extract the netlist from your layout, convert it to a SchematicConfiguration object, then call this method on it:
https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/circuitviz.py#L460

tvt173, 2023-05-18:
Alternatively, create a new `SchematicEditor` object, and populate it with the info from the extracted netlist, and call `.visualize()`. This would probably be better if you intend to edit and save your schematic (placements, etc.)-- i.e. make it read/write, and not read-only

closed 2023-10-21 by github-actions[bot]