gdsfactory/gdsfactory #1666: 1nm snapping issues

joamatab, 2023-05-18:
1nm snapping issues

snapping can happen

see `gdsfactory.components.mzi`

[image]

```python
    c = mzi(cross_section="strip")
    c = gf.routing.add_fiber_array(c)
    c.show()
```

joamatab, 2023-05-22:
this can be fixed with `write_gds(transform_invalid_references=True)` but also creates issues in some other layouts

tvt173, 2023-05-22:
@joamatab, as discussed before, please make sure you are not doing any explicit snapping anywhere (i.e. snapping ports or lengths) when you have that flag set to `True`, as it might conversely cause snapping issues in that case. You can quickly test this out by temporarily commenting out the body of those functions so that they do nothing, and rebuilding your layout... I'd be interested to see if there are still any snapping errors after taking these precautions...

closed 2023-05-22 by joamatab