gdsfactory/gdsfactory #1681: `port_symmetries` on `gtidy3d.write_sparameters` seem broken

Jan-David-Black, 2023-05-21:
**Describe the bug**
When using port symmetries it seems to execute the wrong simulations and not copy over the s-parameters correctly

**To Reproduce**
```python
c = gf.components.mmi2x2()
symmetry = {
  "o1": {
    "s11": ["s22","s33","s44"], "s21": ["s12","s34","s43"], "s31": ["s13","s24","s42"], "s41": ["s14","s23","s32"],
  }
}
df = gt.write_sparameters(c, overwrite=True, port_symmetries=symmetry, is_3d=False, verbose=True)
```

**Expected behavior**
Execute a single simulation (not three) and copy the s-parameters as advised by the dict.

**Suggested fix**
I will open a PR

Jan-David-Black, 2023-05-21:
Additionally it seems to break with the old s-parameter notation. So either we should autoconvert from the old notation to the new one (assuming fundamental mode), or we should fix the docs...

joamatab, 2023-05-22:
it seems still has issues

[image]

lucas-flexcompute, 2023-05-23:
The problem seems to be the name of the ports. The names coming from the component are `'o1', 'o2'`, but the [symmetry definition](https://github.com/gdsfactory/gdsfactory/blob/23a8aa14c4df1c1253f7983196bec9d812bd0ee2/gdsfactory/simulation/port_symmetries.py#L1) uses another naming scheme, so [this if clause](https://github.com/gdsfactory/gdsfactory/blob/23a8aa14c4df1c1253f7983196bec9d812bd0ee2/gdsfactory/simulation/gtidy3d/write_sparameters.py#L181) is never true.

Jan-David-Black, 2023-05-23:
I had been "fixing" the issue according to the [doc string] (https://github.com/gdsfactory/gdsfactory/blob/23a8aa14c4df1c1253f7983196bec9d812bd0ee2/gdsfactory/simulation/gtidy3d/write_sparameters.py#L87). There the port names are used as keys. The format specifying the excite port actually makes sense to me, so we would have to adjust the symmetry definitions. What do you think @joamatab?

Jan-David-Black, 2023-05-23:
- [ ]  And we should add some tests to avoid breakage in the future :)

joamatab, 2023-05-23:
we have tests for the tidy3d plugin and are failing at the moment

Jan-David-Black, 2023-05-23:
Interesting... I'll have to check why the tests passed on my machine...

Jan-David-Black, 2023-05-23:
Regarding the top level keys of the symmetry dicts: do we stick to using the excitation port?

joamatab, 2023-05-23:
when you work on a fork that has no credentials for tidy3d it skips those tests

I think i found a fix, see PR

PR #1683 (2023-05-21): Repair symmetries for Tidy3D
fixes #1681
commit 56dcb311 (2023-05-21): Should fix #1681
closed 2023-05-23 by joamatab