gdsfactory/gdsfactory #1693: tidy3d 2D simulations issues

joamatab, 2023-05-23:
The following 2D waveguide has Sparameters larger than 1

```python
import numpy as np

import gdsfactory as gf
import gdsfactory.simulation.gtidy3d as gt

if __name__ == "__main__":
    # test_sparameters_straight()

    overwrite = False
    overwrite = True
    c = gf.components.straight(length=3)
    sp = gt.write_sparameters_1x1(c, overwrite=overwrite, is_3d=False, run=True)

    # Check reasonable reflection/transmission
    assert 1 > np.abs(sp["o1@0,o2@0"]).min() > 0.8, np.abs(sp["o1@0,o2@0"]).min()
    assert 0 < np.abs(sp["o1@0,o1@0"]).max() < 0.1, np.abs(sp["o1@0,o1@0"]).max()

```

@lucas-flexcompute
@momchil-flex 
@tylerflex

tylerflex, 2023-05-23:
how much larger? if just a few % over (eg 1.05) it is probably due to resolution not being fine enough.

joamatab, 2023-05-23:
it goes from 4 to 0.6 

while for 3D is between 0.99 and 0.999

tylerflex, 2023-05-23:
hm, sounds like normalization. @shashwat-sh if I remember you did some work on the normalization in 2D? Do we have a test of how that's working? If possible @joamatab you could share the json file of the simulation?

momchil-flex, 2023-05-23:
First thing to check is probably the field decay, if that's ok then it may be source normalization.

tylerflex, 2023-05-23:
I just get ```KeyError: 'o1@0,o2@0'```

tylerflex, 2023-05-23:
I dont think the script runs any simulation for me

shashwat-sh, 2023-05-23:
Which version of Tidy3D was being used here? There was a bug in the mode solver in Tidy3D v2.2 which will be fixed in an upcoming patch. Might be related to that.

shashwat-sh, 2023-05-23:
@joamatab  Looking at the simulation json, it looks like the simulation has zero size along `z`, so it is 2D in the `xy` plane. However, the boundary conditions are defined as PML in `x` and `z`, and periodic in `y`. It should probably be PML in `x` and `y`, and periodic in `z`?

joamatab, 2023-05-23:
we are using 2.1.1

i had originally made PML in x,y and periodic in z and had the same issues

joamatab, 2023-05-24:
using 

```python
    boundary_spec = td.BoundarySpec.all_sides(
            boundary=td.Periodic()
        )

In [3]: np.abs(sp["o2@0,o1@0"]).max()
Out[3]: 4.330214980574785

In [4]: np.abs(sp["o2@0,o1@0"]).min()
Out[4]: 0.4450051467717772

```

and with the values you suggest the numbers look better but are still far from 100% transmission

```

        boundary_spec = boundary_spec or td.BoundarySpec(
            x=td.Boundary.pml(),
            y=td.Boundary.periodic(),
            z=td.Boundary.pml(),
        )

In [1]: np.abs(sp["o2@0,o1@0"]).max()
Out[1]: 0.7073590652292768

In [2]: np.abs(sp["o2@0,o1@0"]).min()
Out[2]: 0.6330496139477119

```

closed 2023-05-24 by joamatab