gdsfactory/gdsfactory #1708: all angle router does not get consistent names

joamatab, 2023-05-29:
all angle router does not get consistent names

https://github.com/gdsfactory/gdsfactory/actions/runs/5113997431/jobs/9193803571

@tvt173

tvt173, 2023-05-29:
since when, @joamatab ? do you know what change triggered this?

joamatab, 2023-05-29:
not sure what triggered it, the new difftest catches this issues now as it detects cell name changes that are not at the top level

tvt173, 2023-05-29:
sure, i can take a look... i'd suspect it has to do with floating point numbers passed as part of the transformation... 

in this case, i suppose it would be good if we can get the names to come out the same consistently. but i'm wondering if, in general, we want difftest to be this picky... usually i'm happy if the top-level cell name + XOR is correct, and I would look at a _netlist regression_ for any changes like this to cell names or parameters. i.e. sometimes i might make changes to a cell hierarchy and want to confirm that the layout still comes out the same (XOR-clean) independent of changes to hierarchy/logical structure. i haven't tried the new difftest yet, but I think it would be nice if we are still able to confirm this

tvt173, 2023-06-20:
hi @joamatab , I believe this one was fixed by #1781

closed 2023-06-21 by joamatab