gdsfactory/gdsfactory #1713: Default installation `python -m pip install gdsfactory` requires kfactory

sebastian-goeldi, 2023-05-30:
**Describe the bug**
The default installation now requires kfactory due to the default import of `difftest`. This results in a package that when installed freshly doesn't work and crashes on import.

**To Reproduce**
`python -m pip install gdsfactory` in a fresh environment

**Expected behavior**
The package should install in a working condition by default

**Suggested fix**
Either add kfactory as a dependency or make the import of difftest not default and add a warning when trying to import without having kfactory installed

joamatab, 2023-05-30:
Hi Sebastian,

In the difftest I added try, except
https://github.com/gdsfactory/gdsfactory/pull/1712

How does that sounds?

closed 2023-05-30 by joamatab

sebastian-goeldi, 2023-05-30:
LGTM :+1:

PR #1714 (2023-05-30): fix basic installer
fixes #1713