gdsfactory/gdsfactory #1720: Cadence plugin

95-martin-orion, 2023-06-01:
**Is your feature request related to a problem? Please describe.**
Some of my team would like to use Cadence for graphical design and GDSFactory for scripting. Example flows include:
- Creating a component in Cadence and importing it in GDSFactory
- Creating a component in GDSFactory and modifying it in Cadence
- Arbitrary nesting of GDSFactory components in Cadence components and vice versa

**Describe the solution you'd like**
A plugin similar to the klayout integration, using [SKILL Bridge](https://github.com/unihd-cag/skillbridge) for python-to-SKILL interop with Cadence.

95-martin-orion, 2023-06-01:
First of all: is this feasible? I recognize this is potentially a very large request, and there are likely many hidden issues to be uncovered during development (e.g. curve representation, similar to #1466). On the flip side, this could be hugely beneficial to e.g. Cadence users looking to migrate their workflow.

joamatab, 2023-06-01:
Hi Orion, we are looking for contributors to this type of plugin

analog designers are asking to generate Verilog-A code from gdsfactory. VerilogA code is for simulating Analog. We are looking at [VLSIR](https://github.com/Vlsir) for that @growly  
@dan-fritchman @curtisma  as well as [OpenVAF](https://github.com/pascalkuthe/OpenVAF) @pascalkuthe 
@dkramnik @sunjin-choi @jtambasco 
 
- Creating a component in Cadence and importing it in GDSFactory. Why not use GDS? I do this all the time.
- Creating a component in GDSFactory and modifying it in Cadence. Why not use GDS for the geometry and verilogA for SpectreRF simulation?
- Arbitrary nesting of GDSFactory components in Cadence components and vice versa? Once you import a cell in Cadence you can take it from there

tvt173, 2023-06-02:
this looks like an interesting project also, which may be useful...
https://github.com/jonpry/Pill

95-martin-orion, 2023-06-02:
> * Creating a component in Cadence and importing it in GDSFactory. Why not use GDS? I do this all the time.
> * Creating a component in GDSFactory and modifying it in Cadence. Why not use GDS for the geometry {...}

I'm not the primary user for this flow, but my understanding is that they create PCells in either tool, then port them to the other tool and want the parameters to still be accessible / mutable. To my knowledge this is not something you can achieve with just GDS.

daquintero, 2023-06-17:
For simulation, worth mentioning that `cocotb` which a pretty big python verilog simulation and verification project already has VerilogA Spectre integration https://docs.cocotb.org/en/stable/examples.html#mixed-signal-analog-digital and adding it in a digital design flow might be reasonable for design verification anyway.

joamatab, 2023-06-17:
Nice Dario 

PR for this would be great 

@growly

daquintero, 2023-06-18:
I have already been working on a gdsfactory plugin to integrate it, that is still green which is why I hadn't mentioned yet, https://github.com/daquintero/piel . The idea is to provide a set of interconnection functions between gdsfactory core and the existing [IIC-OSIC-TOOLS](https://github.com/iic-jku/iic-osic-tools) workflow. I thought about integrating it directly into gdsfactory, ie through this eg local branch https://github.com/daquintero/gdsfactory/tree/openlane_integration .

The problem is an environment one. The configuration for OS microelectronic design tools is a lot more involved than for just GDSFactory, and there is a lot more interaction between the design files and environment flags. It might add a bit too much overhead to add it as a dependency directly into gdsfactory than as an external plugin package like `piel`. There are a lot more dependencies and a docker environment configuration that might be too much overhead for a default gdsfactory installation for most users. What do you think? I'll make another issue to discuss as this is a bit deviating from this main Cadence plugin issue.

joamatab, 2023-09-12:
We Added in gplugins the option to export to Spice in Spectre RF syntax

https://github.com/gdsfactory/gplugins/compare/v0.4.0...v0.5.0

95-martin-orion, 2023-09-15:
Does the solution above allow Cadence / SPICE access to the component parameters defined in GDSFactory? IIUC the raw GDS generated from a GDSFactory design does not contain the parameters used to generate that design.

joamatab, 2023-09-16:
The interface will only work for spice simulations and you need to define your own mapping 

Feel free to move this to gplugins