gdsfactory/gdsfactory #1722: 1nm snapping issue

joamatab, 2023-06-02:
[image]
[image]

1nm snapping issue in `test_get_bundle_all_angle.py` with gaps. which it already had before too

joamatab, 2023-06-02:
@sebastian-goeldi 

Would this happen in kfactory?

sebastian-goeldi, 2023-06-02:
Hmm, hard to say. Does it happen on both sides? If your component is long enough and the port wide enough it can happen, on one side of the port only though. So in that case you would be able to first fix it with the min_space fixer. We could theoretically use something that picks the two points of each port and make sure when flattening that they overlap. Maybe it would also make sense in the non-90° case (in kfactory), to treat a port as an edge instead of a point with a rotation. that way you can basically fit both edges on top of each other.

joamatab, 2023-06-02:
it only happens on one side

danielpeace, 2023-06-05:
Having a similar issue when constructing a bent directional coupler.

[image]

[image]

joamatab, 2023-06-05:
Hi Daniel, 

could you share the code to reproduce?

have you tried flattening the bends?

@tvt173

danielpeace, 2023-06-05:
Hi @joamatab 

I haven't tried flattening, how do you recommend flattening the bends?

Thanks!

```
import gdsfactory as gf
import numpy as np

gf.config.rich_output()
PDK = gf.generic_tech.get_generic_pdk()
PDK.activate()

# Broadband SOI curved / straight directional coupler based on: https://doi.org/10.1038/s41598-017-07618-6

@gf.cell
def bent_coupler_half(
    g: float = 0.200,
    Rc: float = 26,
    Ls: float = 1.8,
    Lc: float = 8.6,
    w1: float = 0.400,
    w2: float = 0.400,
    Lx: float = 10,
):
    # Rc = R1-(w1+g)/2 = R2 + (w2 +g)/2
    # Lc = 2 * alpha * Rc

    R1 = Rc + (w1 + g) / 2
    R2 = Rc - (w2 + g) / 2
    alpha = round(np.rad2deg(Lc / (2 * Rc)), 4)
    beta = alpha

    c = gf.Component("bent_coupler_half")

    outer_bend = c << gf.components.bend_circular(
        radius=R1,
        angle=gf.snap.snap_to_grid(-alpha),
        cross_section=gf.cross_section.cross_section(width=w1),
        bend=gf.components.bend_circular(),
        with_bbox=False,
    )

    inner_bend = c << gf.components.bend_circular(
        radius=R2,
        angle=-alpha,
        cross_section=gf.cross_section.cross_section(width=w2),
        bend=gf.components.bend_circular(),
        with_bbox=True,
    )

    outer_bend.movey((w1 + g) / 2)
    inner_bend.movey(-(w2 + g) / 2)

    outer_straight = c << gf.components.straight(
        length=Lc, cross_section=gf.cross_section.cross_section(width=w1), npoints=100
    )

    inner_straight = c << gf.components.straight(
        length=Lc, cross_section=gf.cross_section.cross_section(width=w2), npoints=100
    )

    outer_straight.connect(port="o1", destination=outer_bend.ports["o2"])
    inner_straight.connect(port="o1", destination=inner_bend.ports["o2"])

    outer_exit_bend = c << gf.components.bend_circular(
        radius=R2,
        angle=alpha,
        cross_section=gf.cross_section.cross_section(width=w1),
        bend=gf.components.bend_circular(),
        with_bbox=True,
    )

    inner_exit_bend_down = c << gf.components.bend_circular(
        radius=R2,
        angle=-beta,
        cross_section=gf.cross_section.cross_section(width=w2),
        bend=gf.components.bend_circular(),
        with_bbox=True,
    )

    inner_exit_bend_up = c << gf.components.bend_circular(
        radius=R2,
        angle=alpha + beta,
        cross_section=gf.cross_section.cross_section(width=w2),
        bend=gf.components.bend_circular(),
        with_bbox=True,
    )

    outer_exit_bend.connect(port="o1", destination=outer_straight.ports["o2"])
    inner_exit_bend_down.connect(port="o1", destination=inner_straight.ports["o2"])
    inner_exit_bend_up.connect(port="o1", destination=inner_exit_bend_down.ports["o2"])

    inner_exit_straight = c << gf.components.straight(
        length=Lx, cross_section=gf.cross_section.cross_section(width=w2), npoints=100
    )
    inner_exit_straight.connect(port="o1", destination=inner_exit_bend_up.ports["o2"])

    outer_exit_straight = c << gf.components.straight(
        length=abs(inner_exit_straight.ports["o2"].x - outer_exit_bend.ports["o2"].x),
        cross_section=gf.cross_section.cross_section(width=w1),
        npoints=100,
    )
    outer_exit_straight.connect(port="o1", destination=outer_exit_bend.ports["o2"])

    c.add_port("o1", port=outer_bend.ports["o1"])
    c.add_port("o2", port=inner_bend.ports["o1"])
    c.add_port("o3", port=outer_exit_straight.ports["o2"])
    c.add_port("o4", port=inner_exit_straight.ports["o2"])

    return c

@gf.cell
def bent_coupler(
    g: float = 0.200,
    Rc: float = 26,
    Ls: float = 1.8,
    Lc: float = 8.6,
    w1: float = 0.400,
    w2: float = 0.400,
    Lx: float = 10,
):
    c = gf.Component("bent_coupler")

    right_half = c << bent_coupler_half(g=g, Rc=Rc, Ls=Ls, Lc=Lc, w1=w1, w2=w2, Lx=Lx)
    left_half = c << bent_coupler_half(g=g, Rc=Rc, Ls=Ls, Lc=Lc, w1=w1, w2=w2, Lx=Lx)

    left_half.mirror_x()
    left_half.connect(port="o1", destination=right_half.ports["o1"])

    c.add_port("o1", port=left_half.ports["o3"])
    c.add_port("o2", port=left_half.ports["o4"])
    c.add_port("o3", port=right_half.ports["o3"])
    c.add_port("o4", port=right_half.ports["o4"])

    return c

if __name__ == "__main__":
    c = gf.Component("top")
    coupler = c << bent_coupler(
        g=0.200, Rc=26, Lc=8.1, Ls=1.8, w1=0.400, w2=0.400, Lx=5
    )
    print(coupler.ports)

    c.write_gds("bent_directional_coupler.gds")
```

joamatab, 2023-06-05:
Thank you for sharing Daniel,

i see two possible fixes:

1. define paths, concatenate them and extrude them
2. grow and shrink the polygons

## 1. define paths, concatenate them and extrude them

```python
import gdsfactory as gf
import numpy as np

gf.config.rich_output()
PDK = gf.generic_tech.get_generic_pdk()
PDK.activate()

@gf.cell
def bent_coupler_half(
    g: float = 0.200,
    Rc: float = 26,
    Ls: float = 1.8,
    Lc: float = 8.6,
    w1: float = 0.400,
    w2: float = 0.400,
    Lx: float = 10,
) -> gf.Component:
    # Rc = R1-(w1+g)/2 = R2 + (w2 +g)/2
    # Lc = 2 * alpha * Rc

    R1 = Rc + (w1 + g) / 2
    R2 = Rc - (w2 + g) / 2
    alpha = round(np.rad2deg(Lc / (2 * Rc)), 4)
    beta = alpha

    c = gf.Component("bent_coupler_half")

    inner_bend = gf.path.arc(
        radius=R2,
        angle=-alpha,
    )

    inner_bend.movey(-(w2 + g) / 2)

    inner_straight = gf.path.straight(length=Lc)

    inner_exit_bend_down = gf.path.arc(
        radius=R2,
        angle=-beta,
    )

    inner_exit_bend_up = gf.path.arc(
        radius=R2,
        angle=alpha + beta,
    )
    inner_path = inner_bend + inner_straight + inner_exit_bend_down + inner_exit_bend_up

    c = inner_path.extrude(cross_section=gf.cross_section.cross_section(width=w2))
    return c

if __name__ == "__main__":
    c = bent_coupler_half()

    c.show()
```

## 2. grow and shrink the polygons

```python
import gdsfactory as gf
[... 130 lines cut ...]

    return c

if __name__ == "__main__":
    # c = gf.Component("top")
    # coupler = c << bent_coupler(
    #     g=0.200, Rc=26, Lc=8.1, Ls=1.8, w1=0.400, w2=0.400, Lx=5
    # )
    # print(coupler.ports)
    nm = 1e-3
    import shapely as sp

    c = bent_coupler_half()
    p = c.get_polygons(as_shapely=True)
    # p2 = p[0] + p[1:]
    pm = sp.polygonize(p)
    d = 10 * nm
    pm = pm.buffer(+d)
    pm = pm.buffer(-d)
    c2 = gf.Component()
    c2.add_polygon(pm, layer=(1, 0))
    c2.add_ports(c.ports)
    c2.show(show_ports=True)

```

danielpeace, 2023-06-06:
Both those look perfect, thanks!

joamatab, 2023-06-06:
Actually we got a fix that makes your original code work as well without any gaps

https://github.com/gdsfactory/gdsfactory/pull/1737

closed 2023-06-06 by joamatab