gdsfactory/gdsfactory #1731: gdsfactory/gdsfactory/pdk.py PDK class validate_layers method

alibillalhammoud, 2023-06-05:
This issue deals with the validate_layers method of the PDK class on line 240 in gdsfactory/gdsfactory/pdk.py. This method checks if the PDK contains the layers in the list "required_layers". The "required_layers" variable is not something the user can pass directly to the validate_layers method, instead it is a global variable defined on line 41 of pdk.py

The solution I'd like:  
make required_layers an argument of the validate_layers method so we can pass the layers we require and check if the PDK has these layers.

alibillalhammoud, 2023-06-08:
This is a small update and I can make a PR with the changes if this is a good idea. Please provide feedback. Thoughts anyone?

tvt173, 2023-06-08:
Hi @alibillalhammoud . So I think the `layers_required` is typically used to enforce the layers required in a PDK for all features of gdsfactory to work. And my understanding is that you want to customize this for your PDK? What is your use case/motivation?

alibillalhammoud, 2023-06-09:
Hello @tvt173, 
I was looking through the Pdk class and saw the `validate_layers` method perfectly suited for my use case, and also that `layers_required` was empty (so `validate_layers` is not being utilized effectively in the current version). I also did not see where `layers_required` was being accessed or modified outside `validate_layers`. I have now looked more at the history of `validate_layers` and saw that it [was originally a pydantic validator](https://github.com/gdsfactory/gdsfactory/blob/4b12572aeebd7bce500da75246d53a29c1fa261f/gdsfactory/pdk.py#L60), but was modified [for the reasons described here](https://github.com/gdsfactory/gdsfactory/blob/9c96b6595d843bef5aaa758cd17ebf559385cf21/CHANGELOG.md?plain=1#L5). If we were to make a change as below, we would not change the interface of the method (compatible with existing call to `validate_layers`) while allowing for more generic layer validation functionality.
```
def validate_layers(self, layers_required: Optional[List[Layer]] = []):
        for layer in layers_required:
            if layer not in self.layers:
                raise ValueError(f"{layer!r} not in Pdk.layers {list(self.layers.keys())}")
```

Below is more context on what I am working on...

I am part of the [OpenFASOC](https://github.com/idea-fasoc/OpenFASOC) team. We make open-source analog circuit generators. There are some small analog cells used in OpenFASOC that we have had to manually design (see [header](https://github.com/idea-fasoc/OpenFASOC/blob/main/openfasoc/generators/temp-sense-gen/blocks/sky130hd/gds/HEADER.gds)). Moving forward, we want to automate design of small blocks needed. We are using GDSFactory for this.

**Motivation for this issue**: we are looking for areas where it may be useful to everyone in the open-source silicon community if we directly contribute to the tools used in our flow rather than applying new features to our codebase. Layer validation is a generic feature which may be suitable for GDSFactory. 

The generators we are developing are python functions (ComponentFactory). Our goal is for the functions to accept a Pdk object so we have significant reuse of code once we define the Pdk class for a new technology. To improve reuse of code we defined a dictionary generic layer names (strings) which maps to the actual PDK layers. With this approach we would like the ComponentFactory to verify at runtime that generic layers it will access actually map to something in the PDK object, so we would want a method like 'validate_layers'. 

Please let me know if this is a suitable suggestion,
thanks

tvt173, 2023-06-09:
sorry, i was speaking from memory and didn't take the time to check the codebase... you seem to be right that `layers_required` is empty right now. i think when this was implemented it was intended for that to be populated by special layers like `TEXT` and `PORT` which are expected to exist by some functions in gdsfactory, right @joamatab ?

otherwise, i have no issue with your suggestion. i would suggest making a MR for it so we can review. i do only have one small comment on your code snippet though... it's bad practice to have a mutable object as a default argument for a function. I would suggest setting the default to `None` and in the case of `None`, maybe check against that globally defined `layers_required` list? (assuming we actually add some layers to it...)

alibillalhammoud, 2023-06-10:
tvt173, Thank you for this information about mutable object default args; I will remember for future use. 
See new PR with this feature.

joamatab, 2023-06-10:
PR looks great, thank you Ali!

closed 2023-06-10 by joamatab