gdsfactory/gdsfactory #1734: design region not visible

kdotom, 2023-06-05:
The adjoint example on this page:
https://gdsfactory.github.io/gdsfactory/notebooks/meep/001_meep_sparameters.html

produces the final design in the simulation object, but when plotting with
`optimized_component.plot()`
the design region is not output. The same behavior applies when writing a gds.

All lines of code have been used exactly. Is there any extra action that is necessary? / Any geometry type conversion, etc.?

drinwater, 2023-06-05:
Hi,

`optimized_component.plot()` is supposed to plot the gdsfactory Component. It's only the optimized Component.

-SkandanC

kdotom, 2023-06-05:
Thanks for the note.
`opt.sim.plot2D()`
plots the full device
[image]
this confirms that the simulations worked.

However,
`optimized_component.plot()`
appears to just plot the waveguides
[image]

kdotom, 2023-06-08:
Just found it. Thresholding issue. I interpreted `threshold_offset_from_max` as the normalized weights. This appears to be in terms of epsilon. Leaving a note in case it's helpful. Thanks!

closed 2023-06-08 by kdotom