gdsfactory/gdsfactory #1744: bring 3.7 testing back?

joamatab, 2023-06-08:
Some people are still using python3.9 and we have some new dependencies (kfactory) that require python >=3.10

We need to bring the python3.7 tests back

Possible Solutions:

- rewrite difftest and use klayout only
- migrate kfactory to python3.7 

@SkandanC

SkandanC, 2023-06-08:
Difftest isn't the inly thing using kfactoyr though, the fill and spacing fixers are all using kfactory, and are kfavtory utilities.

Kfacory to py3.7, perhaps, the issue is match case is used in some places, which is inly py3.10 syntax

Would things break for pre 3.10 if we kept tests to 3.10? I don't think there'd be any issues for any plugins, or for gdstk.

joamatab, 2023-06-08:
What do you think of  the simplest solution of running some bare bones tests?

https://github.com/gdsfactory/gdsfactory/pull/1745

SkandanC, 2023-06-08:
Sure, sounds good.

#1745<https://github.com/gdsfactory/gdsfactory/pull/1745>

closed 2023-06-08 by joamatab