gdsfactory/gdsfactory #1771: Slack for GDSFactory?

daquintero, 2023-06-13:
The Efabless open source electronic people have a Slack for users and developers where good developer interactions and announcements can be had:
e.g. https://join.slack.com/t/open-source-silicon/shared_invite/zt-1x5aw19vo-P9yOdTPxARDvqYy~qB6i1A

So I wonder if in `gdsfactory` we have something like this or would benefit from it? Happy to either make a `gdsfactory` channel on their `open-source-silicon` slack or our setup our own slack and transfer ownership to you? Lots of admin but just an idea in case it would benefit the community as in the electronics side.

joamatab, 2023-06-13:
Hi Dario,

we have a [gitter channel](https://matrix.to/#/#gdsfactory-dev_community:gitter.im) which is the open source version of slack :)

daquintero, 2023-06-13:
Ahh cheers Joaquin, yeah that's great! Didn't find it before

closed 2023-06-13 by daquintero

joamatab, 2023-06-13:
I also created a slack channel for gdsfactory in the [open-source silicon dev](https://join.slack.com/share/enQtNTQwMTg0NTI3Nzk5MS01NTZmMDJhNGM1MGVhNTY4ODMyY2ZjZDRhODllMGMyYTdmNmYzOTEyYjdmOTBkMGRkNjc2MjZhOTg4NzI0MGFk)