gdsfactory/gdsfactory #1779: fix schematic editor

joamatab, 2023-06-16:
how can we fix schematic editor?

See docs
https://github.com/gdsfactory/gdsfactory/actions/runs/5283520410/jobs/9559878377

How can we integrate it as part of `gf web`?

joamatab, 2023-06-21:
fixed by https://github.com/gdsfactory/gdsfactory/pull/1788

joamatab, 2023-06-21:
For some reason it still does not work on my vscode

@SkandanC

closed 2023-06-21 by joamatab

tvt173, 2023-06-21:
@joamatab which extension are you using to run these jupytext notebooks in vscode? I can't get the notebooks to run in vscode at all

joamatab, 2023-06-21:
good point,

we can add this jupytext to the developer docs

https://gdsfactory.github.io/gdsfactory/layout.html

tvt173, 2023-06-21:
hmm that's the one i was trying to, but it didn't work for me. i'll have to try again later. works for me in jupyterhub though