gdsfactory/gdsfactory #1791: Better docs for developer installation

tvt173, 2023-06-20:
**Is your feature request related to a problem? Please describe.**
I had to re-do my developer installation recently, and I found the documentation lacking and hard to navigate. Here are my observations.

1. The developer installation docs do not appear anywhere on the nav bar. They are, however, on the main page, but they are so far down it was not obvious to me that's where to find them.
[image]
2. The process for setting up tests (fetching test data, etc.) does not seem to be documented anywhere. There is a section on the tests in the `Contributing` section, but it does not explain how to set them up.
https://gdsfactory.github.io/gdsfactory/contribution.html#tests
3. When running on Windows, I found that the pre-commit hooks would crash when trying to install. Eventually running `mamba install ipykernel` into my conda environment seemed to fix the issue, but I'm not sure if that's conclusively the recommended fix or why.

**Describe the solution you'd like**
1. Installation should be listed on the main navbar of the docs.
2. The process for setting up and running tests, I would argue, should be included in the `Developer Installation` section. The `Contributing` section should link back to the `Developer Installation` section as a prerequisite.
3. Currently in the developer installation procedure, it is documented that you need to install gdstk via mamba. I think ipykernel also should be installed via mamba in that step (but it would be good if someone else on Windows can confirm that...).

joamatab, 2023-06-21:
Hi Troy, good point,

how about this?
https://github.com/gdsfactory/gdsfactory/pull/1793

joamatab, 2023-06-23:
Hi Troy, closing this

I hope it's good now
https://gdsfactory.github.io/gdsfactory/developer.html

closed 2023-06-23 by joamatab