gdsfactory/gdsfactory #1838: Via port inconsistency

EliDoris, 2023-07-07:
In trying to instantiate a `ring_double_heater` I noticed that naming of returned ports is different if the function is supplied a `via_stack` specification or a `via_stack_slot` specification. `via_stack` results in electrical ports `e1` and `e2` whereas `via_stack_slot` gets suffixes resulting in ports such as `e1e4` and `e2e4`

It seems like this has something to do with the difference in how the functions call the `compass` component, but I'm not exactly sure. I would expect that both via specifications should just result in `e1` and `e2` as electrical ports

joamatab, 2023-07-07:
Thank  you for reporting this!

is seems to be a bug on c.auto_rename_ports()

there is a quick fix of removing on `ring_double_heater.py`

```
c.auto_rename_ports()

```

Here is the code to reproduce

```
    from gdsfactory.components import ring_double_heater 

    c1 = ring_double_heater(via_stack="via_stack")
    c1.pprint_ports()

    c2 = ring_double_heater(via_stack="via_stack_slot")
    c2.pprint_ports()

```

Im trying to understand why this is happening

joamatab, 2023-07-07:
the issue is that one had ports of type 'placement' and the other one 'electrical' and the auto_rename ports was not renaming the ports with type placement

PR #1844 (2023-07-08): Fix via stack port orientations
- fix Component.auto_rename ports to also auto-rename ports of port_type='placement'. Fixes #1838 
- default port_type = 'placement' 'electrical' for most cases
- add all ports in straight heater. Adding prefixes `l` for left and `r` for right

@simbilod 
@EliDoris
closed 2023-07-08 by joamatab