gdsfactory/gdsfactory #1842: get_component_hash() default path in Windows not writable

dominauta, 2023-07-07:
**Describe the bug**
To get tidy3D simulation S parameters, a default path for the output file must be provided together with the component. Otherwise the default windows temp folder (C:\Users\XXXX\AppData\Local\Temp\" is seen as not writable by the get_component_hash method and an error is thrown:

``` bash
OSError: Error opening output file.
[GDSTK] Unable to open GDSII file for output

```
This is a known problem with tempfile.NamedTemporaryFile() method.

**To Reproduce**
``` python
import gdsfactory.simulation as sim
import gdsfactory.simulation.gtidy3d as gt
import gdsfactory as gf

gf.config.rich_output()
PDK = gf.generic_tech.get_generic_pdk()
PDK.activate()

c = gf.components.mmi1x2()
s = gt.get_simulation(c, is_3d=True)
fig = gt.plot_simulation(s)
sp = gt.write_sparameters_1x1(c)
```

**Expected behavior**
The temporary GDS file should be created to obtain the hash and ideally deleted to avoid generating unnecesary garbage.

**Suggested fix**
``` python
def get_component_hash(component: gf.Component) -> str:
    with tempfile.NamedTemporaryFile(delete=False) as file:
        path = Path(file.name)
        dirpath = path.parent
        dirpath.mkdir(exist_ok=True, parents=True)
        component.write_gds(path)
        myhash  = hashlib.md5(file.read()).hexdigest()
        file.close()
        import os
        os.unlink(path)
        return myhash
```

joamatab, 2023-07-15:
Can a windows user help with this?

@SkandanC 
@simbilod 
@tvt173

dominauta, 2023-07-16:
I believe it’s the other way around. I provided the solution for windows
but I’m not sure if it will affect Linux machines behavior

halilcuma, 2023-07-16:
I'm encountering exactly the same problem when running the script to obtain S-parameters for a complement.

I'm getting the error: OSError: Error opening output file.

So, what is the suggested solution to tackle this issue? (Please see the attached image for more information.)
[image]
[image]
 @joamatab @dominauta

joamatab, 2023-07-22:
Thank you David!

Here is a simple fix, i bought a windows laptop and tested it

https://github.com/gdsfactory/gdsfactory/pull/1898

closed 2023-07-22 by joamatab

joamatab, 2023-07-22:
Can you `pip install gdfactory --upgrade` and try again?

it should be fixed on the new 6.114.1 release?