gdsfactory/gdsfactory #1863: Advanced packing

simbilod, 2023-07-14:
**Is your feature request related to a problem? Please describe.**

`gdsfactory.pack` is great, but uses component bbox'es. This is inefficient if the components have lots of dead space, which often happens when pads and grating couplers need some minimal clearance. It also does not consider possible component transformations, like rotations.

**Describe the solution you'd like**

Using solutions to the irregular bin packing problem instead would help for non-rectangular bounding boxes. Finding a solution to leverage "holes" in components would also be great (recursive bin packing?)

@jimohyusuf @EliDoris

Makkiy, 2023-10-29:
Let me investigate this & perhaps contribute some advanced packing algorithm implementations.

joamatab, 2024-04-10:
i think this would be a possible better algorithm than the current rectpack

https://github.com/albert-espin/knapsack-packing

joamatab, 2025-04-07:
[This](https://github.com/JeroenGar/sparrow) would be great to integrate

sebastian-goeldi, 2026-09-14:
would recommend this https://github.com/JeroenGar/jagua-rs