gdsfactory/gdsfactory #1886: Create a 3D meshing tutorial

nikosavola, 2023-07-20:
**Is your feature request related to a problem? Please describe.**
There could be a tutorial showcasing 3D meshing. Currently there are ones for both types of 2D meshes

[image]

**Describe the solution you'd like**
Make tutorial for 3D meshing.

Ideally there would be a nice interactive widget.

joamatab, 2023-07-20:
Nice, PRs welcome for this :)

simbilod, 2023-07-20:
[Meshwell ](https://simbilod.github.io/meshwell/)(the 3D meshing backend) has docs but it would be nice to document the gdsfactory interface to it

joamatab, 2023-08-03:
i'll close this, and we can open it again on https://github.com/gdsfactory/gplugins

closed 2023-08-03 by joamatab