gdsfactory/gdsfactory #1914: Use explicit `.j2` endings for component definition YAMLs

nikosavola, 2023-07-27:
**Is your feature request related to a problem? Please describe.**
Continuation to #1908.

Use explicit `.j2` endings for component definition YAMLs with Jinja2 features (or maybe for all?).
This enables Jinja2 linting (#1911) and syntax highlighting in IDEs.

**Describe the solution you'd like**
* Rename example component definitions with Jinja2 features to `.pic.yml.j2` or something similar ending with `.j2`
* Check what changes are needed in code to possibly parse both raw `.yml` and `.j2` endings

nikosavola, 2023-08-03:
I guess in the meeting this was deferred. Instead the issue could be about hacking the current Jinja2 linter to check for custom file-endings. Honestly, this should be a feature in the package so that can be done upstream.

joamatab, 2023-08-03:
Sounds good to me, feel free to send a PR

nikosavola, 2023-09-06:
https://github.com/aristanetworks/j2lint/pull/68 now implements different file extensions

PR #2058 (2023-09-06): Use j2lint for gdsfactory YAML files
This PR extends the allowed filetypes for j2lint to also lint gdsfactory YAML files

Closes #1914
closed 2023-09-06 by joamatab