gdsfactory/gdsfactory #1925: remove all references of a specific component from a parent component

jimohyusuf, 2023-07-31:
**Is your feature request related to a problem? Please describe.**

If there isn't already a way to do this, it might be handy to have a method that allows a component to selectively remove some user-specified references that it contains and return a new component. Currently, the `component.extract()` function allows removing layers, perhaps there could be a similar function for extracting only references of specified components?

**Describe the solution you'd like**

Something like this would be nice:

`c = gf.c.ring_double()`
`c2 = c.remove_refs(["coupler_ring"])`

and after this, I am left with only two straights without the ring.

Also, if there isn't already a way of doing this, is there a suggestion on how it can be implemented? I will be happy to give the implementation a shot.

Thanks!

tvt173, 2023-07-31:
something like this should work, if you want to filter purely on cell name

```python
refs_to_remove = [r for r in c.references if r.parent.name == "coupler_ring"]
for rr in refs_to_remove:
    c.references.remove(rr)
```

or something like this if you want to filter based on the function used to make the subcells
```python
refs_to_remove = [r for r in c.references if hasattr(r.parent.settings, "function_name") and r.parent.settings.function_name == "coupler_ring"]
for rr in refs_to_remove:
    c.references.remove(rr)
```

jimohyusuf, 2023-07-31:
Hi Troy, thanks for your comment! 

So with the first solution proposed, I do print `print(c.references)` before and after the remove operations and the references list did not change. That's fine though as maybe coupler_ring isn't the parent. Instead, I do `c.references.remove(c.references[0])` to remove the first reference which in this case was one of the coupler_rings and when I print the references list, I can see that it is gone. However, on plotting the component or showing it in Klayout, it doesn't seem to have changed at all (all polygons are still present) even when I remove all the references manually as described above.

What could perhaps be going wrong? Thanks!

joamatab, 2023-08-01:
yes, something is not right

```python

import gdsfactory as gf

c = gf.components.ring_double()
c = c.copy()

for i, ref in enumerate(c.references):
    print(i, ref.name)
    if ref.name.startswith("coupler"):
        print(f"deleting {ref.name}")
        c.remove(ref)

c.show()

```

only one coupler reference seems to be removed

[image]

sebastian-goeldi, 2023-08-01:
This might be related to how gdstk is handling the references.

It is generally speaking dangerous to remove elements while iterating over it.

It is almost always the smart decision to just store the elements to remove temporarily and then remove them in a second loop. This even applies to pure python lists.

jimohyusuf, 2023-08-01:
Indeed, the problem seems to be with the list iteration. I was able to get it to work with the code below:

```Python
import gdsfactory as gf

c = gf.components.ring_double()
c =  c.copy()

refs2remove = []
for i, ref in enumerate(c.references):
    if ref.name.startswith("coupler"):
        print(i, ref.name)
        refs2remove.append(ref)

pprint(c.references)

for ref in refs2remove:
    c.remove(ref)

pprint(c.references)

c.plot_matplotlib()
```

[image]

Thanks all! Would be nice to have this as a component method.

tvt173, 2023-08-01:
it seems like a pretty niche thing to add to the Component API if you ask me. maybe as a separate utility function? also note that there are also some limitations to the way you have it implemented

1. Using `ref.name` will only work if the reference was named automatically. You are also allowed to name the refs manually, which would potentially break this assumption. Using the `function_name` setting from my first suggestion should be more robust, at least for cells created with gdsfactory `@cell` functions. I think in some cases (i.e. a cell imported from GDS) this might break also.
2. This will only remove references from the top level of hierarchy. You would need to make a recursive function or something of that sort if you want this to work hierarchically. In that case, you'd also need to be careful to use and substitute copies of the child cells at each level.

joamatab, 2023-08-01:
yes, im pretty sure there is a better way, if you are trying to do this few questions come up.

why not use a hierarchical cell where you only add what you need?

Lets say you have

```python
def ring():
  c = gf.Component()
  ...
  return c

```

```python
def ring_with_pads():
  c = gf.Component()
  ring = c << ring()
  pads = c<< pads()
  ...
  

```

Also even the code you posted had the waveguides messed, see in red where they should be

[image]

jimohyusuf, 2023-08-01:
Thanks for both of your comments!

I definitely agree that this could be a niche thing and could create some problems if not properly implemented. Thanks for all your help and I am happy that I now know how to do it in cases where I need something like that and I know exactly what I am doing.

Also, the scenario I needed this for wasn't for the ring and that was just used as an example. Rather, I have a custom component from which I would like to extract only the metal layers (for parasitic extraction) and the silox openings on the pads. Using `c.extract(layers=(metal1, metal2, silox))`, I also grab the silox layer on other components' refs that have silox layers not just the pads. My thought was just to remove the other component(s) and then do the layer extraction like `c.remove_refs("unwanted_component(s)").extract(layers=(metal1, metal2, silox))`. I ended up just making a new component (before asking here) without the "trouble-causing" components like Joaquin suggested but I imagined a hypothetical situation where I would like to do different layer extractions from the same component and creating a new component every time to avoid issues might not be the most desirable approach. Perhaps that's something anyone rarely has to do in which case there is no reason to worry about it.

Thanks again and I would consider this issue solved from my end.

closed 2023-08-11 by joamatab