gdsfactory/gdsfactory #1955: gplugins/ tidy3d/ materials.py

ChihYuLee, 2023-08-07:
I tried to run this: 
c = gf.components.mzi_phase_shifter()
s = gt.get_simulation(c)
fig = gt.plot_simulation(s)

But it has errors:
ValueError: Material 'tin' not in ['si', 'sio2', 'sin']

I couldn't find where the tin comes from, probably the heater on the shifter. 
Anyway, I am wondering if we have more flexibility of using other materials included in tidy3d library or even add new medium while using component from gf and run simulation on tidy3d.

joamatab, 2023-08-07:
Hi, ChihYuLee,

i wouldn't simulate an MZI in tidy3d, as it can be quite big, i would only simulate the couplers

See how to model the MZI phase shifter here
https://gdsfactory.github.io/ubc/notebooks/14_sax_tidy3d.html

ChihYuLee, 2023-08-07:
Dear Joaquin,

Thank you for your reply.
How about tidy3d on some less computationally-intensive devices with some
customized materials?
I guess now the gplugins issue is the materials we can use is limited, is
there any trick?
Back to the error codes I got, do you have any solutions beside switching
to sax?

Best regards,
Chih-Yu

joamatab, 2023-08-08:
You need to ensure that your material corresponds with the materials available in the tidy3d material library. For further discussions or concerns, I'd suggest redirecting this matter to the [tidy3d repository](https://github.com/flexcompute/tidy3d). Additionally, it's worth noting that the heater from the generic PDK is composed of TiN, which is commonly used for heaters, and does not seem to be available in tidy3d material database

closed 2023-08-11 by joamatab