gdsfactory/gdsfactory #1959: Add Component along a path

joamatab, 2023-08-07:
[Add Component along a path](https://heitzmann.github.io/gdstk/how-tos.html#points-along-a-path)

it would be nice to add examples on how to do this

95-martin-orion, 2023-08-07:
I wrote a function for this once, but don't currently have time to clean it up for a PR:

<details><summary>Code</summary>

```python
@gf.cell
def along_path(
    p: gf.Path,
    feature: gf.Component,
    spacing: float,
    padding: float,
) -> gf.Component:
    """Returns Component containing many copies of `feature` along `p`.

    Places as many copies of `feature` along each segment of `p` as possible
    under the given constraints. `spacing` is always followed precisely, but
    actual `padding` may exceed the provided value to place features evenly.

    Args:
        p: Path to place features along.
        feature: Component to repeat along the path. The unrotated version of
            this object should be oriented for placement on a horizontal line.
        spacing: distance between feature placements.
        padding: minimum distance from corner to first feature.
    """
    from shapely.geometry import LineString

    c = gf.Component()

    path_linestr = LineString(p.points)
    cum_dist = 0
    for i, start_pt in enumerate(path_linestr.coords[:-1]):
        end_pt = path_linestr.coords[i + 1]
        segment_length = LineString([start_pt, end_pt]).length
        num_features = int((segment_length - (padding * 2)) // spacing)
        if num_features < 1:
            continue

        feature_array = c << gf.components.array(
            feature,
            spacing=(spacing, 0),
            columns=num_features,
            rows=1,
        )
        offset = (segment_length - ((num_features - 1) * spacing)) / 2
        angle = np.rad2deg(np.arctan2(end_pt[1] - start_pt[1], end_pt[0] - start_pt[0]))
        # Offset, rotate around new origin, then move to line segment.
        feature_array.move((offset, 0)).rotate(angle).move((start_pt[0], start_pt[1]))
        cum_dist += segment_length
    return c
```

</details>

Behavior:

```
 <-p+k-> <-s-> <-s-> <-s-> <-s-> <-p+k->
o-------x-----x-----x-----x-----x-------o
```
* The maximum number of features allowed by `spacing` and `padding` will be added to the path. Spacing (feature-to-feature distance) is rigidly enforced, but padding (feature-to-endpoint) will adjust such that the features are centered on the path.
* This implementation treats each path segment separately, which may not always be desirable. For example, unless padding is very small, this usually means that features don't get placed on curves.
* While this was sufficient for my use case, it is likely suboptimal.

joamatab, 2023-08-07:
looks great, thanks for sharing Orion!

tvt173, 2023-08-08:
the all-angle router supports the concept of custom connectors. you could define a custom connector to do something like this ([as shown here](https://gdsfactory.github.io/gdsfactory/notebooks/04_routing_non_manhattan.html#bends-and-connectors)), and then just specify it as part of your routes. however, the limitation would be that it would only be applicable on the straight sections (not the bends)

joamatab, 2023-08-11:
Thank you for the code Orion and the PR Nicholas!

closed 2023-08-11 by joamatab