gdsfactory/gdsfactory #1986: Class methods clickable or expose more information in docs

nikosavola, 2023-08-15:
**Is your feature request related to a problem? Please describe.**
Class methods in the generated API documentation don't show all information nor are they clickable for seeing the full definition

For example, the `get_netlist` method for https://gdsfactory.github.io/gdsfactory/_autosummary/gdsfactory.Component.html only shows this much information
[image]

while the actual docstrings contain this much
[image]

**Describe the solution you'd like**
Make the methods clickable for full definition and maybe show more information like argument types.

drinwater, 2023-08-16:
I like what we do with kfactory documentations for the API, where have the entire source code for each class and function, but this is in mkdocs, which ius different from jupyter. https://gdsfactory.github.io/kfactory/reference/kfactory/kcell/#:~:text=information%20in%20um-,Source%20code%20in%20src/kfactory/,-kcell.py

-SkandanC

joamatab, 2023-08-16:
yes, i like it more too,

@HelgeGehring  has also been looking at this

https://myst-tools.org/docs/mystjs

tvt173, 2023-08-16:
i think maybe that adding the `:recursive:` directive, as shown in this [SO post](https://stackoverflow.com/a/62613202/701803) might fix it...

[image]

[image]

joamatab, 2023-08-16:
i tried that but didn't work

sebastian-goeldi, 2023-08-17:
The problem here is that you just specify **kwargs as an argument but then use specific arguments (not only kwargs which is misleading itself already). If you want the full arguments, you need to write all the args to the definition of the function or put the definition below the kwargs entry of the doc string args

sebastian-goeldi, 2023-08-17:
> The problem here is that you just specify **kwargs as an argument but then use specific arguments (not only kwargs which is misleading itself already). If you want the full arguments, you need to [...]

For most docstring collectors, they will look for args in the function definition and match to the docstring and not vice-versa afaik

nikosavola, 2023-08-18:
I think the type information would also be useful. That is currently not visible for the ones where they are defined.

joamatab, 2023-09-08:
We have been experimenting with myst here

https://github.com/joamatab/gdsfactory

https://joamatab.github.io/gdsfactory 

once we figure out how to better document the API we can consider it

@HelgeGehring

closed 2023-11-15 by github-actions[bot]

nikosavola, 2023-12-07:
This issue still persists