gdsfactory/gdsfactory #1997: get_netlist_flat: Map each connection to its corresponding polygons

das-dias, 2023-08-21:
Hello! I don't think I quite understood the operation of get_netlist_flat; more specifically its instance and connection mapping. Is there a way of mapping each connection (between two ports) to the corresponding polygons that implement that connection?

tvt173, 2023-08-21:
Hi Diogo,

Have you tried the `report_pathlengths` utility in gplugins?
https://gdsfactory.github.io/gplugins/notebooks/path_length_analysis.html

I think it does roughly what you need, or should at least get you most of the way there 🙂

tvt173, 2023-08-22:
so, if we trace what the `get_edge_based_route_attr_graph()` function is actually doing, it
- iterates recursively through the extracted netlist of the circuit
- finds components with `route_info` attached to them. here we trust the user/pdk owner to have decorated these components properly... it is virtual data added to the components manually and indepently from polygon creation (usually by reporting the length of the extruded path in the component, but can be set in a more custom manner if necessary)
- reconstructs the hierarchical circuit as a graph, where component ports are nodes and edges are the paths between them

then when you run `get_paths()` you are extracting out the paths in the graph, which condenses sequences of connected routes into composite paths.

all that is to say, none of it actually uses polygons _or_ paths. it relies on virtual data annotated by the component author. however, the info for each component along each path gets annotated back onto the edges of the graph, so you can traverse the graph yourself and extract the polygons and/or paths out of the components as necessary. if you call `get_polygons()` from the components traversed and the `include_paths` flag is set to True (default), then it should return all polygons and paths (represented as polygons), optionally for the specific layer you need

das-dias, 2023-08-22:
Perfect, I can work with that!

closed 2024-01-01 by github-actions[bot]