gdsfactory/gdsfactory #2004: Corner + U case in auto router

tvt173, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
There is one case of `get_bundle()` where one must make a bundle by either
A. Turning before crossing the final port and making a U back towards it (if there is enough room)
B.  Turning after crossing the final port and making a U back towards it

**Option A:**
[image]

**Option B:**
[image]

This is currently not supported, and will throw a RouteWarning. However, it would be nice to support it.

**Describe the solution you'd like**
Ideally, test if there is enough room for option A and use it if possible. Otherwise option B.

The case shown above represents two of the skipped `tests\routing\test_port_ordering.py` cases, with `make_bundle(angle=270, reverse_ports=True,)`

joamatab, 2023-08-22:
yes, it would be great to support it

@SkandanC 
@sebastian-goeldi 
@damienbonneau

joamatab, 2024-10-15:
Closing this as fixed

[image]

closed 2024-10-15 by joamatab

sebastian-goeldi, 2024-10-15:
I think it's currently not possible to choose the other option though. But I feel generally this is the better solution anyway, as this "avoids" having to route around the whole component. I believe though the other piece errors out even if another way would be available. But I feel that this is a corner case where the current solution is "good enough"