gdsfactory/gdsfactory #2009: Customizable difftest settings/exclusions

tvt173, 2023-08-23:
**Is your feature request related to a problem? Please describe.**
The updated `difftest()` function drives me crazy. It's way too sensitive. I'd like to introduce options that would allow the user to make it ignore certain changes, such as
1. Changes to cell names
2. Changes to labels
3. Changes to certain layers might be a nice feature also (i.e. if you want to check for changes to labels in general, but exclude certain label layers).

An implicit part of this change would be to make the test distinguish between sliver and label changes. Currently when reporting to the console, it states that changes are one or the other but it's hard to tell which.

**Describe the solution you'd like**
difftest has options
- exclude_cell_names: bool
- exclude_labels: bool
- exclude_layers: list[LayerSpec] | None

It also reports in the console whether a change was due to sliver or label specifically, not grouping the two together.

joamatab, 2023-08-24:
I like this idea, the default is too sensitive

How can we configure the klayout difftool to ignore some of those cases?

Shall we ignore in gdsfactory.config.difftest_settings?

@SkandanC 
@sebastian-goeldi

sebastian-goeldi, 2023-08-24:
Just hook to the right functions and ignore the other changes & layers you aren't interested in. https://klayout.de/doc-qt5/code/class_LayoutDiff.html

joamatab, 2023-12-07:
I think we fixed it Troy, feel free to reopen