gdsfactory/gdsfactory #2025: prevent dangerous manual cell naming

joamatab, 2023-08-26:
deprecate  `gf.Component(name='whatever')`

I've seen many people of setting the manual name, which can cause duplicated cell names

here are some options:

1. We deprecate the behaviour and raise a warning, saying that you need to use `@gf.cell` for naming cells
2. We append an incremental number. This can cause problems when merging cells generated at different run times
3. We append a `uuid` unique identifier

What do you think?

@tvt173 
@HelgeGehring 
@sebastian-goeldi

HelgeGehring, 2023-08-26:
I think best would be to just use functions.

That way people are directly used to it and there no extra thing to learn once they do their own conponents

joamatab, 2023-08-26:
something like this

```
Use @cell decorator for auto-naming. Component(name) will be removed soon. Example:

@gf.cell
def my_component():
    c = gf.Component()
    c.add_ref(gf.components.straight())
    return c

```

tvt173, 2023-08-27:
sounds good. I'm not 100% clear if this is what you mean @HelgeGehring , but I think it would be nice also to add a `component.rename()` function that allows the user to manually rename in a safe way (checking first for duplicates in the cache and then registering).

i find that sometimes doing things like this is necessary, i.e. if you want to generate a batch of DOEs or golden cells with a specific naming convention from a CSV or config file

sebastian-goeldi, 2023-08-27:
I think 2. is a good solution. But iirc this is only a problem on saving? Or does gdstk not allow this during runtime anymore?

If it does, the solution can be pretty easy: Instead of appending during runtime, you do it during saving, i.e. in `gds_write` you add (blatantly coping what klayout does) $1/$2/$3... and so on to duplicates.

uuid have the disadvantage that it's not deterministic, meaning your cells get a (partially) random name.

Another way, which klayout as well does (partially), is to check on creation in the layout map. In gdstk, I think, this would amount to a global storage of components or at minimum component names. This would also allow you to check for other shenanigans, namely error such as calling two different functions with same name and footprint. I would argue even two functions with the same name shouldn't be allowed.

How that could work is by storing function names globally and when `@cell` is called, it would throw an error if more than one function gets called, ever. E.g. a dict per name could be used and store name/memroy id in dict of `{function_name: {memory_id1: 0, memory_id2: 1, ...}`. If a function is called its memroy_id value gets set to 1. As soon as the dict values get >1, you would throw an error. Additonally to help debug, throw a warning when more than 1 function is registered withthe same name.

joamatab, 2023-08-27:
i tried implementing both 1. and 2.

1. https://github.com/gdsfactory/gdsfactory/pull/2026
2. https://github.com/gdsfactory/gdsfactory/pull/2027

I think name2 provides a better user experience, but im getting difftest errors on the tests that I can't figure out why they are happening

joamatab, 2023-08-27:
i found the issue, 

we were calling gf.clear_cache() on the tests,

maybe we can reset the counter in that case

joamatab, 2023-08-27:
After implementing 2. I can see it has a much nicer user experience

We can also merge 1 as well

PR #2027 (2023-08-27): Name counter
- implement name counter for avoiding name conflicts when manually naming cells or importing from different gds files

fixes #2025 

@tvt173 
@sebastian-goeldi
closed 2023-08-27 by joamatab