gdsfactory/gdsfactory #2039: issue on "fill_rectangle"

wuzhonghan, 2023-08-30:
when I test the fill_rectangle_custom()  with margin=5, the code is below:

**mzi = gf.components.mzi()
c = gf.Component("component_with_fill")
layers = [(1, 0)]
c << fill_rectangle_custom(
    mzi,
    gf.components.rectangle(),
    avoid_layers=layers,
    margin=5,
    spacing=(4, 2),
    bbox=mzi.bbox,
)
c << mzi
c.show(show_ports=True)**

[image]
I find in some region the margin is less than 5, how can I fix this?

joamatab, 2023-08-30:
thanks for reporting this

why not using the tiling [processor](https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html#dummy-fill-tiling)?

@SkandanC

SkandanC, 2023-08-30:
Hi,

I think tilin gprocdessor is only useful for large layouts, not for a single mzi. @sebastian_goeldi knows more.

joamatab, 2023-08-30:
yes, but since they are using GDS anyway to interface with kfactory 

however im not sure where there is issue coming from

@sebastian-goeldi

sebastian-goeldi, 2023-08-30:
Shouldn't be a problem to use the tiling processor for even small things. There is some overhead to calculate the string and set up the processor, but that's in the order of 10-100ms would be my guess. So as long as you don't do it 10'000 times, no problem.

wuzhonghan, 2023-08-31:
thanks for reply,

I will try tiling processor

closed 2023-09-22 by joamatab