gdsfactory/gdsfactory #2045: simpler sort ports

joamatab, 2023-08-30:
we recently introduced `enforce_port_ordering` flag but we already had the `sort_ports` flag with served for the same purpose

How about we use `sort_ports` only?

@tvt173

tvt173, 2023-08-30:
Yes, I still maintain the view that `enforce_port_ordering` should _never_ be false. If we get rid of the flag and use only the "True" case in all the code, it will make things simpler. 

To be clear though, they _don't_ serve the same purpose. `sort_ports` specifies whether or not to sort ports on one side of the bundle to prevent routes from crossing, etc. `enforce_port_ordering` prevents the error where connections form between the _wrong_ ports, creating essentially a different circuit than what was specified (assuming the intention is that ports 0,1,2,... of ports1 should connect to ports 0,1,2... of ports2). This is the issue detailed here: https://github.com/gdsfactory/gdsfactory/issues/1992

Currently if `enforce_port_ordering` is set to False, it is saying instead "connect ports 0,1,2,... of ports1 to _any one of ports_ 0,1,2... of ports2".

joamatab, 2023-08-31:
I see, I closed the PR

how about naming it keep_port_order ?

tvt173, 2023-08-31:
how about something like `swappable_ports`? I think this is common term in LVS for example. you can specify that pins are swappable, which is basically the same concept as here.

in this case, however, the boolean would need to be flipped. `swappable_ports = True` would be equivalent to `enforce_port_order = False`

closed 2023-08-31 by joamatab