gdsfactory/gdsfactory #2080: Types used inconsistently, start using mypy

nikosavola, 2023-09-12:
**Is your feature request related to a problem? Please describe.**
The codebase has many places where typing is used inconsistently. For example, instead of `Layer`s, sometimes the bare definition of `tuple[int, int]` is used.
https://github.com/gdsfactory/gdsfactory/blob/314204879f48a11bc9610f8ce4e5174de009936b/gdsfactory/port.py#L517

https://github.com/gdsfactory/gdsfactory/blob/314204879f48a11bc9610f8ce4e5174de009936b/gdsfactory/component.py#L1488

Additionally, some functions allow inputs that are either single elements or lists or tuples (or even any sequences). For example,
https://github.com/gdsfactory/gdsfactory/blob/314204879f48a11bc9610f8ce4e5174de009936b/gdsfactory/cross_section.py#L1335
This is fine itself but makes it easy to miss edge-cases if the code is not doing something like this:
https://github.com/gdsfactory/gdsfactory/blob/314204879f48a11bc9610f8ce4e5174de009936b/gdsfactory/cross_section.py#L1445-L1446
(also note that the argument types don't allow a list but we're checking for one here)

Particularly these types of error-prone sections could be easily caught and prevented with mypy.
Enforcing mypy would help prevent errors from similar sections emerging for new code. In addition, the real allowed argument types would have to be written explicitly. This shouldn't be too much trouble if [abstract base classes](https://docs.python.org/3/library/collections.abc.html) are used.

**Describe the solution you'd like**

Enforce mypy in CI with some minimal config at first.

drinwater, 2023-09-12:
Hi Niko,

We have tred usign mypy a year ago, only to run into thousands of issues :)

You can take a look at prevous issues on mypy here.

In kfactory, we use I think ruff?, but the codebase is a lot smaller and newer than gdsfactory, so its easy to fix new mypy issues.

nikosavola, 2023-10-25:
Instead of mypy, type checking could be done with pyright, for which a pre-commit hook also exists https://github.com/microsoft/pyright/blob/main/docs/ci-integration.md

joamatab, 2023-10-25:
i added it here and ended up commenting as there also are thousands of errors :)

https://github.com/gdsfactory/gdsfactory/compare/update_pre-commit?expand=1

nikosavola, 2023-10-25:
Yeah there are loads of errors, most stem from confusing stuff like Components and ComponentFactories etc. I think

95-martin-orion, 2023-11-14:
I'd like to echo the push for this. While I recognize the difficulty of setting up typechecking initially is considerable, the value added for both users and developers cannot be overstated. Type annotations enable:

- Variable type-tracking in IDEs, allowing code hints to work in any context. Without this, users are forced to check the documentation (or even the source) to find what functions a variable supports.
- Early issue detection in gdsfactory PRs, even without a full test suite. Type checks will catch out-of-order function arguments, misuse of list types (list / tuple / ndarray), and other common bugs.

The mypy site provides [helpful instructions](https://mypy.readthedocs.io/en/stable/existing_code.html) for enabling type checking on an existing codebase, and a specific example of incrementally adding type checking can be seen [here](https://medium.com/alan/python-typing-with-mypy-progressive-type-checking-on-a-large-code-base-74e13356bd3a).

nikosavola, 2023-11-14:
Incrementally adding type checking sounds like a great idea to advance this issue. I might take a look where we could enable it in the near future. 

I'm leaning towards using pyright because of a better language server experience for the user, although mypy has the benefit of having a Finnish lead developer :smile:.

nikosavola, 2023-11-15:
There seems to be a lot of trouble from @gf.cell and partials. For example, `gdsfactory/read/import_gds.py` has two errors that seem mysterious:
```
/home/nikosavola/gdsfactory/gdsfactory/read/import_gds.py:160:14 - error: Expected 1 more positional argument (reportGeneralTypeIssues)
  /home/nikosavola/gdsfactory/gdsfactory/read/import_gds.py:164:20 - error: Argument of type "Literal['a.gds']" cannot be assigned to parameter of type "(**P@cell) -> ((gdspath: (str | Path), cellname: (str | None) = None, gdsdir: (str | Path | None) = None, read_metadata: bool = False, keep_name_short: bool = False, unique_names: bool = True, max_name_length: int = 250, **kwargs: Unknown) -> Component)"
```
Seems pyright nor mypy can infer the argument types from a function with a partialed decorator, or the types have to be something convoluted like here https://stackoverflow.com/questions/69027497/mypy-errors-for-decorator-w-arguments.
https://github.com/python/mypy/issues/1484 may be related as well.

joamatab, 2023-11-15:
Maybe we can target that for the next major release 8.0.0, which is at the moment tentatively using the klayout backend

Sebastian made the `cell` decorator work with mypy

nikosavola, 2023-11-15:
Sounds like a good idea to do in conjunction with the move to kfactory

sebastian-goeldi, 2023-11-21:
You have to be very careful with the declaration of the type of the decorator. Especially if you want it to be usable both with and without kwargs (as a call or without call). But to boil it down:

It needs to take a `Callable[..., Component]` and it also returns that. So, you have to be very careful to make the footprint to match that behavior.

I think you could largely copy the return type and structure from kfactory and make it work in gdsfactory if necessary.

95-martin-orion, 2024-02-01:
Bumping this issue to remove staleness label. Is this still planned for kfactory / v8.0 ?

sebastian-goeldi, 2024-02-01:
@95-martin-orion kfactory already uses mypy and passes every test for it. I think klayout doesn't ideally export it everywhere (or the classes in kfactory are too complex to resolve in some cases), but it passes the mypy pre-commit (only the notebooks are not mypy tested as of now):

https://github.com/gdsfactory/kfactory/actions/runs/7727821283/job/21067171665

Since the code base size of gdsfactory should shrink considerably when switching to kfactory, it should be easier to complete this.

joamatab, 2024-06-10:
Hi Orion,

we reduced a lot of mypy issues in gdsfactory8 but we are still in need of some help fixing the remaining mypy issues

MatthewMckee4, 2024-11-16:
Hi there, I have just forked gdsfactory and will make a pull request, in the coming days, to try to resolve this issue. I was going to make an issue myself but i will just reference this issue. I hope my help is useful and still wanted.

MatthewMckee4, 2024-11-16:
Is there still any interest in adding some linting commit hooks like ruff lint and mypy?

sebastian-goeldi, 2024-11-16:
ruff is there already through pre-commit, mypy only makes sense once it passes, but then ofc yes it will be nice

Not sure you can make it work in every case. In kfactory I had to run mypy with a non-precommit config as otherwise it would freak out

joamatab, 2024-11-17:
Hi Matthew, 

fixing mypy, extending ruff checks or removing some of the exceptions would be great

```
[tool.ruff]
extend-exclude = ['docs/notebooks/']
fix = true
ignore = [
  "B008",  # do not perform function calls in argument defaults
  "B028",  # stacklevel
  "B904",
  "B905",  # `zip()` without an explicit `strict=` parameter
  "C408",  # C408 Unnecessary `dict` call (rewrite as a literal)
  "C901",  # too complex
  "D100",  # public module docstrings
  "D101",  # public class docstrings
  "D102",  # public method docstrings
  "D103",  # public function docstrings
  "D104",  # public package docstrings
  "E501"  # line too long, handled by black
]
select = [
  "B",  # flake8-bugbear
  "C",  # flake8-comprehensions
  "D",  # pydocstyle
  "E",  # pycodestyle errors
  "F",  # pyflakes
  "I",  # isort
  "T10",  # flake8-debugger
  "UP",  # pyupgrade
  "W"  # pycodestyle warnings
]

```

sebastian-goeldi, 2024-11-17:
Not sure whether pep-8 is worth it.

Generally, I agree with favoring snake_case naming, but as gdsfactory often uses physics/math stuff, I wouldn't enable that at this time.

The functions which break this rule are just component functions. I would not consider it important to have them follow the snake_case naming convention. The exception to this would be if you find functions or variables using Pascal case, which I would definitely avoid, as it would suggest that it's a class.

joamatab, 2024-11-17:
yes, we can add them in later PRs, maybe we can ignore the `pep8 naming rule` for now

MatthewMckee4, 2024-11-18:
Thoughts on [ERA](https://docs.astral.sh/ruff/rules/#eradicate-era) ruff rules?

This has one rule which flags commented out code.

sebastian-goeldi, 2024-11-19:
Theoretically, ERA is nice, but I think there are too many `__main__` sections in gdsfactory which we would lose with valid but commented out local tests.

Also, I think the type rewriting broke a bunch of stuff in dependent packages (not all of them are public). Could you indicate in future PRs if you remove or move existing type(alias) variables? Yes I know it's on us for review, but the PRs are quite big due to the refactoring nature, so it's easy to miss.

MatthewMckee4, 2024-11-19:
My apologies, yes i will do that. Do you know what type alias variables created issues?

joamatab, 2024-11-19:
Closing this for now, Thank you Matthew for your nice work!

closed 2024-11-19 by joamatab

sebastian-goeldi, 2024-11-19:
I think it was mainly types that were moved. It's not super critical, but something we have to be aware of.

MatthewMckee4, 2024-11-19:
> Closing this for now, Thank you Matthew for your nice work!

This issue also talks about mypy, and fixing issues raises with that, id there still an interest in getting this done, I have begun working on this in my fork and will continue it out of my own interest. So will make a PR in the coming weeks (there's 2000 errors so will take a while).

As I've been working on fixing some errors, there seems to be a lot that I'm not sure I can fix myself as I don't want to change core functionality and I don't know enough about the package as of now. If there is interest in me working on this, is there an appropriate place I can ask questions about issues I find?

sebastian-goeldi, 2024-11-19:
You can either open new issues (if it's something missing/wrong), open a new discussion about it, or even easier, join the matrix (or if legacy, gitter) channel https://matrix.to/#/!CdusfnfNMqPjRrBRWk:gitter.im

95-martin-orion, 2024-11-19:
> If you want to add shapes like that, use `c.shapes(LAYER.WG).insert(dbox)` etc. It should be considerably faster, as no casting / checking needed in python.

Thanks for the pointer! Performance is definitely a concern for us, but it can be a bit tricky to track all of the different ways to construct component geometry. Adding this and other optimizations to the best-practices guide from #2173 would be incredibly helpful.