gdsfactory/gdsfactory #2089: issue with gf.path.transition() ports position

gmgopikrishnan, 2023-09-15:
Hi
I noticed that when using the transition function to transition from one cross section to the other, the "o1" port always seems to be at the smaller width and the "o2" port seems to be always at the larger width. I noticed this issue since I last updated gds factory which was last week.

**To Reproduce**
_import gdsfactory as gf

WGLayer = (1,0)

Cband_IO = gf.CrossSection(width = 5, offset=0, layer=WGLayer, name="WG", 
                           port_names=("o1", "o2"))
Cband_SM = gf.CrossSection(width = 50, offset=0, layer=WGLayer, name="WG", 
                           port_names=("o1", "o2"))

WGTrans_IOSM = gf.path.transition(cross_section1=Cband_IO, cross_section2=Cband_SM, width_type="linear")
WGTrans_SMIO = gf.path.transition(cross_section1=Cband_SM, cross_section2=Cband_IO, width_type="linear")

c = gf.Component()

WG1Path = gf.Path()
WG1Path.append(gf.path.straight(length=100, npoints=2))
WG1 = c << gf.path.extrude(WG1Path, cross_section=Cband_IO)

WG2Path = gf.Path()
WG2Path.append(gf.path.straight(length=100, npoints=2))
WG2 = c << gf.path.extrude(WG2Path, cross_section=WGTrans_IOSM)

WG3Path = gf.Path()
WG3Path.append(gf.path.straight(length=100, npoints=2))
WG3 = c << gf.path.extrude(WG3Path, cross_section=Cband_SM)

WG4Path = gf.Path()
WG4Path.append(gf.path.straight(length=100, npoints=2))
WG4 = c << gf.path.extrude(WG4Path, cross_section=WGTrans_SMIO)

WG5Path = gf.Path()
WG5Path.append(gf.path.straight(length=100, npoints=2))
WG5 = c << gf.path.extrude(WG5Path, cross_section=Cband_IO)

WG2.connect("o1", destination=WG1.ports["o2"])
WG3.connect("o1", destination=WG2.ports["o2"])
WG4.connect("o1", destination=WG3.ports["o2"])
WG5.connect("o1", destination=WG4.ports["o2"])

c.show(show_ports=False, show_subports=False)
c.write_gds(gdspath="Test.gds", flatten_invalid_refs=True)_

The output is:
[image]

I would have expected the output to be such that the part circled in blue is reversed
[image]

I am not sure if this a bug or an update in the new version of gdsfactory.
Thanks for your help in checking this.

joamatab, 2023-09-21:
the issue is with the names

```python

import gdsfactory as gf

WGLayer = (1,0)

Cband_IO = gf.CrossSection(width = 5, offset=0, layer=WGLayer, name="WG_narrow",
port_names=("o1", "o2"))
Cband_SM = gf.CrossSection(width = 50, offset=0, layer=WGLayer, name="WG_wide",
port_names=("o1", "o2"))

WGTrans_IOSM = gf.path.transition(cross_section1=Cband_IO, cross_section2=Cband_SM, width_type="linear", name='IOSM')
WGTrans_SMIO = gf.path.transition(cross_section1=Cband_SM, cross_section2=Cband_IO, width_type="linear", name='SMIO')

c = gf.Component()

WG1Path = gf.Path()
WG1Path.append(gf.path.straight(length=100, npoints=2))
WG1 = c << gf.path.extrude(WG1Path, cross_section=Cband_IO)

WG2Path = gf.Path()
WG2Path.append(gf.path.straight(length=100, npoints=2))
WG2 = c << gf.path.extrude(WG2Path, cross_section=WGTrans_IOSM)

WG3Path = gf.Path()
WG3Path.append(gf.path.straight(length=100, npoints=2))
WG3 = c << gf.path.extrude(WG3Path, cross_section=Cband_SM)

WG4Path = gf.Path()
WG4Path.append(gf.path.straight(length=100, npoints=2))
WG4 = c << gf.path.extrude(WG4Path, cross_section=WGTrans_SMIO)

WG5Path = gf.Path()
WG5Path.append(gf.path.straight(length=100, npoints=2))
WG5 = c << gf.path.extrude(WG5Path, cross_section=Cband_IO)

WG2.connect("o1", destination=WG1.ports["o2"])
WG3.connect("o1", destination=WG2.ports["o2"])
WG4.connect("o1", destination=WG3.ports["o2"])
WG5.connect("o1", destination=WG4.ports["o2"])

c.show(show_ports=False, show_subports=False)

```

will give you the correct cross_section

[image]

gmgopikrishnan, 2023-09-21:
Hi
Thank you very much for clarifying! Sorry about that.