gdsfactory/gdsfactory #2104: There was an error filling the square, and the generated gds file was blank.

zrzyyds, 2023-09-20:
**Description error**
There was an error filling the square, and the generated gds file was blank.

`    import kfactory as kf
    from gdsfactory.generic_tech import LAYER

    c = gf.Component()
    ring = c << gf.components.coupler_ring(
        # cross_section=gf.cross_section.rib_conformal, radius=20
    )
    # ring = c << gf.components.coupler_ring()
    gdspath = c.write_gds()
    c.show()

    # floorplan = c << gf.components.bbox(ring.bbox, layer=l.FLOORPLAN)
    # gdspath = c.write_gds()

    d = RegionCollection(gdspath)
    d[LAYER.N] = d[LAYER.WG].copy()
    d[LAYER.WG].clear()

    # d[LAYER.SLAB90] += 2  # grow slab by 2um
    # d[LAYER.SLAB90] -= 2  # shrink slab by 2um
    # d[LAYER.SLAB90].smooth(1000)  # smooth slab by 1um points
    # d[LAYER.DEEP_ETCH] = d[LAYER.SLAB90]  # copy layer
    # d[LAYER.SLAB90].clear()  # clear slab150

    # d.write_gds("out.gds", keep_original=True)
    # gf.show("out.gds")
    d.show()

    # d["SLAB150"] = d.WG - d.FLOORPLAN
    fill_cell = d.get_fill(
        d[LAYER.FLOORPLAN] - d[LAYER.WG],
        size=(0.1, 0.1),
        spacing=(0.1, 0.1),
        fill_layers=(d[LAYER.WG],),
    )
    c = d.get_kcell()
    _ = c << fill_cell
    fill_cell.write("fill.gds")`

[image]

joamatab, 2023-09-20:
Hi, 

this is a kfactory issue so i'd recommend you open the issue there

@sebastian-goeldi 
@SkandanC

SkandanC, 2023-09-20:
Hi,

The gdsfactory fill operation using kfactory was written by Simon. so perhaps he kows more about this @Simon ***@***.***>

joamatab, 2023-09-20:
the gdsfactory fill is documented here

https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html#dummy-fill-tiling

I think you are using the old syntax

closed 2023-09-21 by joamatab