gdsfactory/gdsfactory #2118: grid snapping fixes

joamatab, 2023-09-22:
The recent grid snapping issues fixes are causing gaps in non-manhattan connections

You can take a look at `gdsfactory/samples/snap_bends.py`

[image]

@tvt173 
@sebastian-goeldi

sebastian-goeldi, 2023-09-22:
Yeah, I was wondering whether this wouldn't show up. This is the problem if you force ports on grid. That's why I only see the problem being solved in case of kfactory with virtual components and in your case probably selectively add the port snapping when a manhattan component amd not when used for non-manhattan routing/components

closed 2023-09-25 by joamatab

tvt173, 2023-09-28:
yes, snapping ports to grid will inherently break non-manhattan connections like this. the "flatten invalid refs" fix used at the time of writing the GDS only works because the points are unsnapped and maintain their original floating-point precision