gdsfactory/gdsfactory #2134: the bug for CrossSection with bbox

LuYouxi, 2023-09-28:
When I use get_bundle for routing, the bbox option is added to the route. The offset mode of the bbox is different from the previous version. The offset of the bbox is added to the input and output of the arc and the straight.

```
from functools import partial

import gdsfactory as gf
from gdsfactory.cell import cell
from gdsfactory.component import Component
from gdsfactory.generic_tech import get_generic_pdk
from gdsfactory.port import Port

si_section = gf.Section(width=0.45, offset=0, layer="WG", port_names=['o1', 'o2'])
si_wg = gf.CrossSection(
    sections=[si_section], 
    radius=5, 
    bbox_layers=["WGCLAD"], 
    bbox_offsets=[5], 
    auto_widen=True, 
    auto_widen_minimum_length=50, 
    width_wide= 1,
)

c = gf.Component("sample_connect")
mmi1 = c << gf.components.mmi1x2()
mmi2 = c << gf.components.mmi1x2()
mmi2.move((100, 50))
route = gf.routing.get_route(mmi1.ports["o2"], mmi2.ports["o1"], cross_section=si_wg)
c.add(route.references)
c.show()

```
Version 7.8.0
[image]
[image]

Before version 7.8.0
[image]
[image]

joamatab, 2023-09-28:
yes

there was a problem on the previous behavior for bbox of bends that have angles different than 90, 180 and 270

what's the issue with the new behavior?

LuYouxi, 2023-10-03:
> yes
> 
> there was a problem on the previous behavior for bbox of bends that have angles different than 90, 180 and 270
> 
> what's the issue with the new behavior?

hi joamatab, I think the old waveguide bbox behaved better. After all, no additional cladding is required at the junction between the waveguides. At the same time, this will avoid additional DRC errors.

joamatab, 2023-10-03:
I see, thank you for reporting this!

This PR will bring the old behavior for the bends
https://github.com/gdsfactory/gdsfactory/pull/2154

closed 2023-10-03 by joamatab

LuYouxi, 2023-10-04:
> I see, thank you for reporting this!
> 
> This PR will bring the old behavior for the bends and straights #2154

hi, I saw the modification of the straight waveguide and bend waveguide codes, but the offset parameter of bbox did not work, and the waveguide with bbox is still the same as version 7.8.0. I guess the implementation of add_bbox in cross_section should also be modified.