gdsfactory/gdsfactory #2162: Incorrect width doing path extrude

Wsssl97, 2023-10-08:
**Describe the bug**
The component in the generated gds file has incorrect width doing path extrude (e.g. width is set to be 0.415um but the actual width is 0.416um)

**To Reproduce**
p = gf.path.arc(radius=10, angle=90, npoints=100)
s = gf.Section(width=0.415, offset=0, layer=(2, 0))
x = gf.CrossSection(sections=[s])
c = gf.path.extrude(p, x)
c.write_gds("test.gds")
c.show()

**Expected behavior**
Should be 415nm but it's 416nm
Silimar issue happens with 419nm

[image]

**Suggested fix**
How could we fix the bug?

joamatab, 2023-10-08:
yes, that works as expected since we enforced by default ports to be on grid as well as points

if you want 415nm edges you need to do

```python
import gdsfactory as gf

gf.config.enable_off_grid_ports()
p = gf.path.arc(radius=10, angle=90, npoints=100)
s = gf.Section(width=0.415, offset=0, layer=(2, 0), port_names=["o1", "o2"])
x = gf.CrossSection(sections=[s])
c = gf.path.extrude(p, x)
c.show(show_ports=True)

```
but then you are at risk of having 1nm gaps in your layout

[image]

take a look at this tutorial
https://gdsfactory.github.io/gdsfactory/notebooks/04_routing_non_manhattan.html

Wsssl97, 2023-10-08:
Thanks for your reply. However I copied your code it still generates 416nm waveguide

joamatab, 2023-10-08:
yes, because the port is centered at 0,0 or at 10,0

this is the correct drawing,

due to GDS being at 1nm or 5nm grid in some cases this is always going to happen

[image]