gdsfactory/gdsfactory #2166: Excessive warnings importing gds with long cell names

wyywyy23, 2023-10-09:
**Describe the bug**
Excessive warnings raised when importing a gds file with cell names that are too long (>99), even if specifying `keep_name_short=True`

**To Reproduce**
Importing a gds with very long cell names (>99) with:
```python
import gdsfactory as gf
from gdsfactory.name import get_name_short, clean_name

c = gf.import_gds(
        gdspath=path_to_gds,
        cellname=clean_name(get_name_short(very_long_cell_name)),
        keep_name_short=True,
    )
``` 

**Expected behavior**
Gdsfactory should import the cell and give it the short name. It indeed functions like that, however it gives excessive warning complaining `very_long_cell_name is too long. Max length is 99. Renaming to ...`

**Suggested fix**
In [import_gds.py](https://github.com/gdsfactory/gdsfactory/blob/4cfa14cbf95f2f7a284c34301373de84781692f7/gdsfactory/read/import_gds.py#L64-L71) a component is created for every cell (some with very long names) before shortening the names. The constructor of the Component issues those warning.

joamatab, 2023-10-09:
can you change the CONF at the top of your script?

```
import gdsfactory as gf

gf.CONF.max_name_length = 200

c = gf.import_gds(
        gdspath=path_to_gds,
    )
```

wyywyy23, 2023-10-09:
> can you change the CONF at the top of your script?
> 
> ```
> import gdsfactory as gf
> 
> gf.CONF.max_name_length = 200
> 
> 
> c = gf.import_gds(
>         gdspath=path_to_gds,
>     )
> ```

Sure, and that does suppress the warning.

Just thought that since the `get_name_short()` method actually renames the cell w.r.t. the `max_name_length`, whether it is 99 or 200 or anything, it would make more sense if we don't need to manually increase the `max_name_length` just for getting rid of the warning.

Anyway not an urgent issue and thanks for the hard work and prompt reply!

joamatab, 2023-10-09:
this should fix the issue
https://github.com/gdsfactory/gdsfactory/pull/2167

i increased the limit to 250 characters 
names longer than 250 character crash gdstk

@lucas-flexcompute

closed 2023-10-09 by joamatab