gdsfactory/gdsfactory #2188: Understand the capability of get_route

zhengqigao, 2023-10-16:
**Describe the bug**
I am using gdsfactory to conduct some study on photonic circuit layout. I test the ```get_route``` function with the following example and find it fails with an error. It seems that the routing algorithm fails because of too many iterations? I tried ```with_sbend = True ``` or ```with_sbend = False```. Am I missing anything?

**To Reproduce**

```
import gdsfactory as gf
from gdsfactory.config import CONF
from gdsfactory.generic_tech import get_generic_pdk

PDK = get_generic_pdk()
CONF.display_type = 'matplotlib' # 'matplotlib'  # 'kweb'
PDK.activate()
c = gf.Component(name="my_fist_component")
mmi1 = c.add_ref(gf.components.mmi1x2())
mmi2 = c.add_ref(gf.components.mmi1x2())
mmi1.movey(50)
mmi2.rotate(10)

route = gf.routing.get_route(mmi1.ports["o2"], mmi2.ports["o1"],with_sbend=True)
c.add(route.references)
c.plot()
```

**Expected behavior**
Successfully finishes the routing.

joamatab, 2023-10-17:
for non manhattan routes you need to use the all angle router

```python

import gdsfactory as gf
from gdsfactory.config import CONF
from gdsfactory.generic_tech import get_generic_pdk

PDK = get_generic_pdk()
CONF.display_type = "klayout"  # 'matplotlib'  # 'kweb'
gf.config.enable_off_grid_ports()
PDK.activate()
c = gf.Component(name="my_fist_component")
mmi1 = c.add_ref(gf.components.mmi1x2())
mmi2 = c.add_ref(gf.components.mmi1x2())
mmi1.movey(50)
mmi1.movex(50)
mmi2.rotate(10)

routes = gf.routing.get_bundle_all_angle([mmi1.ports["o2"]], [mmi2.ports["o1"]])
for route in routes:
    c.add(route.references)

c.show()

```

[image]

https://gdsfactory.github.io/gdsfactory/notebooks/04_routing_non_manhattan.html

zhengqigao, 2023-10-17:
Thanks! One last question, how to define Manhattan and non-Manhattan routing?

closed 2023-10-17 by zhengqigao

joamatab, 2023-10-17:
Manhattan needs to have ports with orientation of 0, 90, 180 or 270 degrees