gdsfactory/gdsfactory #2204: How to modify a gds file imported from outside

HUjie321, 2023-10-21:
I have a small problem, which may seem simple, but I am a student and there is no other way to get help, hope to get answered, thank you, I used "import_gds" in my python program to import a gdsII file of a simple inverter layout exported in cadence, how can I import two similar files into one page, and wire or modify them

joamatab, 2023-10-21:
```pytyhon
import gdsfactory as gf

c = gf.Component(top)
c1=gf.import_gds('file1.gds')
c2=gf.import_gds('file2.gds')

# for wiring them you need to define ports
layer_metal = (2, 0)
c1.unlock()
c1.add_port('e1', position=(0,0), width=10, orienation=0, layer=layer_metal)
c2.unlock()
c2.add_port('e1', position=(0,0), width=10, orienation=0, layer=layer_metal)

# and a cross_section for the wiring metal
r1  = c << c1
r2  = c << c2

xs_metal = gf.cross_section.metal3(l, layer=layer_metal)

route = gf.routing.get_route(r1['e1'], r2['e1'], cross_section=xs_metal)
c.add(route.references)
```

HUjie321, 2023-10-22:
Thank you very much for your reply, but I didn't understand the meaning of top, it means gf. Component("name"), the name of the file? and I tried the following code, running without error, but showing an empty Component in klayout. After that I will study the tutorial more seriously, but at the moment this problem does not give me much time, please forgive me for bothering.

import gdsfactory as gf
c = gf.Component("aaa")
c1=gf.import_gds('file1.gds')
c2=gf.import_gds('file2.gds')
c.show()
c.plot()

wwwwwwzx, 2023-10-22:
c is the canvas you put components in and "aaa" is the name of the top cell you will see in Klayout.
You should add c1 and c2 (as component references) to the canvas c (top component) :
```
c << c1
c << c2
```
before the line `c.show()`

HUjie321, 2023-10-23:
Thanks a lot, this really solved my problem