gdsfactory/gdsfactory #2227: Reconsider `get_component` and similar methods

95-martin-orion, 2023-10-25:
**Is your feature request related to a problem? Please describe.**
I've seen this issue crop up in an older code base I'm working with: use of a global accessor to retrieve objects based on their name generates bad code. Most noticeably, it obfuscates dependencies between modules, making it very difficult to prevent import cycles and locate where code is being used.

**Describe the solution you'd like**
There are places where functions like `get_component` are useful - for example, when passing around design files that can only contain descriptions of a component, not the entire python object. However, use of such functions should be restricted to those use cases to prevent bad code patterns.

joamatab, 2023-10-25:
I agree, 

I liked the original way of passing cell functions instead of ComponentSpecs, as it makes the code more readable

@tvt173

tvt173, 2023-10-26:
**Disclaimer:** I'm biased because I'm the one who introduced this, and I still find it valuable.

That said, let me try to recall and summarize all the reasons why we did so

**1. Portability and deferred/inverted dependencies**
  Take the example of a higher level component in the generic PDK that takes another component as a parameter. Let's say that the default subcomponent is an "mmi1x2". In the old way, the default component argument would have been provided as a function. However, let's say that is _not_ the mmi1x2 I use in my platform. I have a different mmi1x2 that should always be used with my PDK. In the past, we wouldn't have been able to use this higher-level component out of the box. We would have had to redefine it somehow, maybe just by creating a new function with a partial, or maybe it would be a bit more involved (i.e. if this issue cascaded down multiple levels).

**2. Simplifying imports and import cycles**
This is really part of the first point, but I'll call it out, since you call it out specifically as well. I don't understand your complaint that it _"makes it very difficult to prevent import cycles"_. Shouldn't the opposite be the case? When using functions, you need to import them before adding them as default values, at the _top level_ of the module. With strings, you are deferring the loading of the subcomponent argument until the component function is actually invoked, so you require no extra import statements. This should _reduce_ import cycle issues.

**2. Serializability**
  We care about serialization in various instances. Component names are created by serializing the arguments + function info. Components also get serialized with their settings when writing out netlists, etc. It is much more straightforward and stable to serialize/deserialize a string or dict than a function. This also helps make it dead-simple to pass a component as an argument in a yaml-defined PIC, as a string or dict. Or through a separate UI, via simple text entry or dropdown.

**3. Dynamic invocation**
Since components can be accessed by name, like `gf.get_component("component_name")`. It makes it very simple to fetch components i.e. which have been registered dynamically, through a spreadsheet maybe, in a consistent way that you can fetch any other component.

Note also that this same pattern has been used for both layers and cross sections. In general I find similar benefits there, so I won't dive into any of those since they're out of scope of your issue. But I do like that there is consistent capability for all three.

I would also argue that if you are a pdk author, you have _no requirement_ to allow `ComponentSpec`s in your function arguments if you don't like them. You can require all your component arguments to be functions or components. But I do find value in the functionality and appreciate having it there.

tvt173, 2023-10-30:
@95-martin-orion I think this depends on expectations, particularly if the expectation is that these functions or features are abstract or concrete. 

the more central and abstract a function is, the more I would argue that it should use the approach of the string-based argument that gets resolved later (or otherwise somehow defer to a configuration-based setting). the more specific and/or concrete the function is, the more it makes sense to directly reference the function. 

for example, in your second example, i would most certainly just use the square function directly also. your first example I would argue is borderline. honestly, I would usually redefine the new component funciton also anyways, especially because I usually like to enforce component naming conventions within a PDK. but I still usually prefer to have the argument as a string because of the simplicity of serialization and imports and flexibility that it brings.

where I care more though are functions which are more central and are used further down. more so than the generic PDK, functions which are defined in the core of gdsfactory itself, which might be embedded within functions that get called by other functions, which get called by other functions etc. often there, we don't wish to pass those parameters all the way up the chain of functions. it is more convenient to have a central configuration which, if set properly, will help the defaults be correct globally. usually, more so than with components, this is important with layers and cross sections. like in the [add_pins functions](https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/add_pins.py#L123). before all of these used to point to concrete layers like (1,0), which for other PDKs might be unusable. for things like these, some means of abstraction is necessary

tvt173, 2023-10-30:
in conclusion to my comment above, i think there is a design tradeoff to either approach, which leads to questions about "which is better for scenario X?" i would tend to lean towards the more abstract and flexible approach in the core library or in functions which are meant to be generic.

closed 2024-02-01 by github-actions[bot]